PhilipsHue/flutter_reactive_ble #10: Restarting/Refreshing in dev mode makes it crash with alreadyInitialized error on iOS

albinhubsch, 2020-02-03:
I have tried this both with the example app and my own app with the same result. Both simulator (iOS) and a device (iPhone 8).

Using the restart command for flutter in dev mode makes this lib crash with the following output
```
[VERBOSE-2:ui_dart_state.cc(157)] Unhandled Exception: PlatformException(alreadyInitialized, already initialized, null)
#0      StandardMethodCodec.decodeEnvelope 
package:flutter/…/services/message_codecs.dart:569
#1      MethodChannel._invokeMethod 
package:flutter/…/services/platform_channel.dart:156
<asynchronous suspension>
#2      MethodChannel.invokeMethod 
package:flutter/…/services/platform_channel.dart:329
#3      FlutterReactiveBle.initialize 
package:flutter_reactive_ble/src/reactive_ble.dart:90
#4      FlutterReactiveBle._trackStatus 
package:flutter_reactive_ble/src/reactive_ble.dart:55
#5      new FlutterReactiveBle._ 
package:flutter_reactive_ble/src/reactive_ble.dart:28
#6      FlutterReactiveBle._sharedInstance 
package:flutter_reactive_ble/src/reactive_ble.dart:23
#7      FlutterReactiveBle._sharedInstance 
package:flutter_reactive_ble/src/reactive_ble.dart:23
#8      new FlutterReactiveBle (package:flutter_reactive_ble/src/reactive_ble.dart:25:3<…>
```

This is especially annoying during **development** as it hangs the whole simulator/device and the app needs to be recompiled and launched again. It does not happen when just using the hot reload functionality.

This shouldn't be a problem in production mode due to the fact that there is no way to "refresh" the app instance. Have not tried if it all applies on Android as well.

remonh87, 2020-02-04:
Thanks for reporting the issue can reproduce the behavior. We are quite strict on reinitalizing the ble client on native level since we want to prevent hanging connections etc. Maybe we are a bit too strict here and for debugging it is indeed very annoying. Will look at the side effects of supporting this use case and come back to you.

werediver, 2020-02-20:
Implemented / fixed in #21. Will be available in the next release.

closed 2020-02-20 by werediver