PhilipsHue/flutter_reactive_ble #12: Allow scanForDevices without service uuid

xvrh, 2020-02-13:
The README for `scanForDevices` states that "the `withService` parameter is required.".

I would like to scan for every BLE devices around. What is the reason to enforce the service filter?

werediver, 2020-02-13:
A single service ID to filter the relevant devices during scanning is required by [scanForDevices()](https://github.com/PhilipsHue/flutter_reactive_ble/blob/d88aa6ae32866cbaead7a9abba1c3a924ce4026a/lib/src/reactive_ble.dart#L226) function indeed, because it is the recommended way to perform scanning (allows for better energy efficiency) and covers majority of the use cases (more or less everything, except for scanner-apps).

If you really want to get as much scanning results as possible (under this restriction), you can scan for for the "Device Information" primary service (see [GATT services](https://www.bluetooth.com/specifications/gatt/services/)) with a short ID 0x180A. This way you'll get most, if not all the advertising devices around.

xvrh, 2020-02-13:
Hi,
Thank you for the answer.
Do you have an example for your recommandation?
I tried this:
```dart
    _ble.scanForDevices(withService: Uuid([0x18, 0x0A])).listen((data) {
      print('Device $data');
    });
```
But I get this unhandled error on Android: `Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scanparameters are not being set")`

werediver, 2020-02-13:
Your code snippet looks fine (another way to construct UUID is `Uuid.parse("18A0")`). You can check the example application (it also performs scanning) and modify it to quickly see the results.

I asked @remonh87 whether the error message rings any bells for him (because it doesn't for me). Let's wait for his feedback.

remonh87, 2020-02-13:
the issue happens because for some reason the scanparameters appears to be empty. This looks like a bug  on our side (I suspect some protobuf conversion going wrong).  Can you share the logcat output?

xvrh, 2020-02-14:
I'm sorry I cannot reproduce the `Scanparameters are not being set` anymore. Sorry for the false alarm.

I launched the "example" application and entered "180a" in the UUID field but I still don't see any devices.

Also here is some random notes about the example app:
- For iOS 13 I need to add to the keys `NSBluetoothAlwaysUsageDescription` and `NSBluetoothPeripheralUsageDescription` in the Info.Plist file
- On Android, by default the bluetooth state was in "unauthorised". To activate it, I had to hack the source code to trigger a requestPermissions.
- The app throws an exception with "Value not in range: 23" when you paste a long uuid in the filter field and press the back keys for a second.

werediver, 2020-02-20:
I ran some tests and have to admit that scanning for the Device Information 0x180A service is useless. Very little devices advertise it. We'll have to change the interface to support scanning with no filter or with multiple service IDs.

xvrh, 2020-02-26:
Would you accept a pull request that change the `scanForDevices` declaration?

```dart
  Stream<DiscoveredDevice> scanForDevices({
    @deprecated Uuid withService,
    List<Uuid> withServices = const [],
    ScanMode scanMode = ScanMode.balanced,
  });
```

remonh87, 2020-02-26:
@xvrh  I already started working on it on https://github.com/PhilipsHue/flutter_reactive_ble/tree/support-blescan-with-empty-filter . Would be great if you can help us out with the manufacturer data.

xvrh, 2020-02-26:
Perfect!

dustin-graham, 2020-03-06:
I believe I need this feature for my application. I'm trying to connect to a device which does not seem to be discoverable unless I remove all filters. I see some good progress in `support-blescan-with-empty-filter`, but no progress on iOS yet. Any estimated ETA on completing this on both platforms? Need any help?

remonh87, 2020-03-06:
I have finished the Android implementation on this branch and will start on the iOS side. I expect we will release this the coming days .

dustin-graham, 2020-03-06:
@remonh87 Awesome.

closed 2020-03-09 by remonh87

remonh87, 2020-03-09:
The functionality is in v2.0 which is available in pub.dev

PR #41 (2020-03-09): Support blescan with empty filter
Fixes #12 
Fixes #23