PhilipsHue/flutter_reactive_ble #13: Add Manufacturer specific data in DiscoveredDevice

xvrh, 2020-02-13:
Could the response of `scanForDevices` contains the Manufacturer specific data?

werediver, 2020-02-13:
I am pretty sure that what you're looking for is [DiscoveredDevice.serviceData](https://github.com/PhilipsHue/flutter_reactive_ble/blob/d88aa6ae32866cbaead7a9abba1c3a924ce4026a/lib/src/model/discovered_device.dart#L28).

xvrh, 2020-02-13:
I was looking for the `SparseArray<byte[]> | getManufacturerSpecificData()"` https://developer.android.com/reference/android/bluetooth/le/AdvertiseData on Android
and https://developer.apple.com/documentation/corebluetooth/cbadvertisementdatamanufacturerdatakey on iOS

Which is different from the service advertisment data.

Both Flutter_blue and flutter_ble_lib expose this:
https://github.com/pauldemarco/flutter_blue/blob/master/lib/src/flutter_blue.dart#L243 
https://github.com/Polidea/FlutterBleLib/blob/develop/lib/scan_result.dart#L36

werediver, 2020-02-13:
You're right, [here](https://github.com/PhilipsHue/flutter_reactive_ble/blob/d88aa6ae32866cbaead7a9abba1c3a924ce4026a/ios/Classes/ReactiveBle/CentralManagerDelegate.swift#L33) we're only extracting `CBAdvertisementDataServiceDataKey` data (and similarly on Android).

This is a valid feature request 👍

xvrh, 2020-02-26:
Would you accept a pull request that add a `manufacturerData` field to `DiscoveredDevices`?

```dart
@immutable
@FunctionalData()
class DiscoveredDevice extends $DiscoveredDevice {
  final String id;
  final String name;
  @CustomEquality(DeepCollectionEquality())
  final Map<Uuid, Uint8List> serviceData;
  
  final Uint8List/*?*/ manufacturerData;

  @CustomEquality(Ignore())
  final int rssi;
}
```

werediver, 2020-02-26:
@xvrh Looks fine. This is not the biggest change needed for the whole feature to work, though.

Please, feel free to open a pull request 👍

PR #32 (2020-03-02): Add manufacturer data in DiscoveredDevice
Fixes #13 

This adds a `Map<int, Uint8List> manufacturerData` field in the `DiscoveredDevice` class.

This exposes the `getManufacturerSpecificData` from Android `ScanRecord` (https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getManufacturerSpecificData())
and the `CBAdvertisementDataManufacturerDataKey` on iOS (https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys)
closed 2020-03-02 by remonh87