PhilipsHue/flutter_reactive_ble #14: Support iOS 13 in the example app

werediver, 2020-02-14:
As reported in [the issue 12](https://github.com/PhilipsHue/flutter_reactive_ble/issues/12#issuecomment-586193161), the example app needs some updates to "Info.plist" to properly work on iOS 13.

Specifically check `NSBluetoothAlwaysUsageDescription` and `NSBluetoothPeripheralUsageDescription` parameters.

diegoveloper, 2020-02-19:
hello, any of the authors of this package are here? I was trying to reach this email `philipshuedev@gmail.com` but I didn't get any answer.

werediver, 2020-02-19:
@diegoveloper The authors are here, me and @remonh87 are the creators and major contributors of the library, but what's your _actual_ question?

Sorry for the "philipshuedev" e-mail. I guess, it's a functional account that we personally do not monitor. I'll see what can be done to improve on this.

diegoveloper, 2020-02-19:
Could you send me your mail?

werediver, 2020-02-19:
@diegoveloper I'd find a separate ticket more appropriate for this discussion. Address your letters to raman.fedaseyeu at gmail.com to reach me.

@remonh87 I'll make sure you're in touch anyway, but check this out.

PR #33 (2020-02-27): Add iOS Bluetooth usage description keys in the sample app
The usage description keys for Bluetooth are required since iOS 13 (I think). Without them, the sample app crashes immediately.

Resolves #14.
closed 2020-02-27 by werediver