PhilipsHue/flutter_reactive_ble #15: Does it work in a separate dart isolate?

espositofulvio, 2020-02-17:
I have a use case where I need to do a scan periodically in a separate isolate and was wondering if flutter_reactive_ble is capable of doing that. Recently flutter_isolate stopped working with all the plugins I'm using so I guess something changed in flutter itself and am looking for something that is capable of running in a different isolate.
Thanks

werediver, 2020-02-17:
@espositofulvio Why are you sure you need a separate isolate for that? The threading on the native side of a plug-in does not depend on the use of isolates on the Dart side of a Flutter program, therefore you cannot move the work done by a plug-in to a background thread by creating new isolates in Dart. But even if the work is being done on the Dart side, it's likely asynchronous (or should be made asynchronous with reasonably sized synchronous chunks), so it's not that likely that isolates are needed.

A BLE scan is definitely not a task that is needed to be moved to a background thread, because what happens inside an application performing scanning is essentially event handling, which is not a lot. All the low level business, the actual BLE protocol implementation is happening on the system side anyway.

espositofulvio, 2020-02-17:
Hi Raman,
the thing is I'm not doing only the ble scan. The isolate is needed because it keeps going when the flutter app is put in background and I need to sync a few BLE devices in the vicinity every 5 minutes. Because of the nature of Dart isolates, I can't pass complex objects so what I do is:

- I start the isolate with the userId
- The isolate runs a function every 5 minutes with timer.periodic
- In the function I make an http call to get the list of devices owned by the user with the above id, scan for devices nearby, monitor a characteristic (which I guess is the equivalent of FlutterBlue setNotifyValue), gather some data from the device and upload to an http service.

This might happen even when the app is not foreground. As of today there isn't yet something like android_alarm_manager that works for both Android and iOS, so I'm stuck with an isolate.

werediver, 2020-02-18:
@espositofulvio Did you find this info about the isolates running when a Flutter app is in background in Flutter documentation? As far as I am aware, the isolates allow you to leverage hardware parallelism (threading), and that's it. As a consequence, you leverage _nothing_ related to running tasks in background on iOS and, I guess, neither on Android (@remonh87 can you check that?). On iOS you'd need to enable some [background modes](https://developer.apple.com/documentation/uikit/app_and_environment/scenes/preparing_your_ui_to_run_in_the_background/about_the_background_execution_sequence) for that, and on Android you may need to implement a service. That's for _reliable_ background processing. A more casual approach would be [extend background execution time](https://developer.apple.com/documentation/uikit/app_and_environment/scenes/preparing_your_ui_to_run_in_the_background/extending_your_app_s_background_execution_time) (to a _limited_ extent) on iOS or just rely on the system not killing the app without a good reason on Android. Threading alone does not enable background processing.

Aside from that, Flutter's plug-in IPC works only on the main thread (on both iOS and Android), which doesn't mean a plug-in can't off-load some work to a background thread, but means the main thread must be involved anyway. And there is no way to to affect a plug-in's threading behavior from Dart.

espositofulvio, 2020-02-18:
@werediver Well, reading this page https://flutter.dev/docs/development/packages-and-plugins/background-processes my understanding is that an isolate can keep running even when the main one is put in background. Also there's a plugin (flutter_isolate) which setup the isolate and does all the wiring for you and allows access to plugins (which unfortunately doesn't work with recent flutter and plugin api v2, but using a custom registrant with a different signature from that generated by flutter itself could make things work again). The thing is, the plugin must be written to work in the background where there's no activity (at least on Android, I don't know much about iOS), and the one I'm using fails to do so referencing the activity from the registrar (which is the main activity in the UI isolate). That's why I was asking if flutter_reactive_ble could work in an isolate where there's no current activity.

werediver, 2020-02-18:
@espositofulvio It appears to me we're hitting some more in-depth Android specificities than I feel comfortable talking about without an extra research (I'm an iOS/Flutter developer with limited Android knowledge). From here on I'd like @remonh87 to engage.

On iOS side it's simpler (one doesn't need to write a plug-in in a very specific way to enable it to work in background, but rather needs to set the application "manifest" properly, declaring one or more suitable background modes; there is some [new iOS 13 API for background tasks](https://developer.apple.com/documentation/backgroundtasks), though, which implies some implementation considerations).

We had no issues using "flutter_reactive_ble" in background (on both iOS and Android), though our use cases are limited, mostly finishing a peripheral's firmware upgrade.

espositofulvio, 2020-02-18:
@werediver A quick explanation. In general it's not that a plugin has to be written in a specific way, but if it references the MainActivity (which is the activity tied to the main UI thread where flutter does all the rendering) then it won't work if run in an isolate. The plugin I'm using has a call to activity.getSystemService() to get the bluetoothadapter and also a background thread calling to activity.runOnUIThread to schedule a callback on the main UI thread, but if it's run from an Isolate, that activity doesn't exists. The first call can actually be simply replaced using a more general context, but the second one can't work. Anyway, I'll give a try replacing flutter_blue with flutter_reactive_ble and see what happens. Thanks!

werediver, 2020-02-18:
@espositofulvio Don't forget Dart/Flutter isolates do not affect Flutter plug-ins execution directly. Regarding the calls you mentioned (`activity.getSystemService()` and `activity.runOnUIThread`), I believe both "flutter_reactive_ble" and [RxAndroidBle](https://github.com/Polidea/RxAndroidBle) (which is used under the hood) are safe, but feel free to run your checks against the source: [android/src/test/kotlin/com/signify/hue/flutterreactiveble](https://github.com/PhilipsHue/flutter_reactive_ble/tree/master/android/src/test/kotlin/com/signify/hue/flutterreactiveble).

I'm closing the ticket for now, because it seems you got enough information to move on. Please, feel free to get back with more details or other specific questions 🙂

closed 2020-02-18 by werediver

Artur110, 2021-02-03:
> Anyway, I'll give a try replacing flutter_blue with flutter_reactive_ble and see what happens. Thanks!

@espositofulvio, can you please tell me if it works?

espositofulvio, 2021-02-03:
@Artur110 I had had a quick stab at it and seemed to work, but then for other reasons (including flutter_blue started working in the background thanks to a couple of pull requests plus using of background_fetch package), I didn't pursue this. I'd still love to replace flutter_blue with flutter_ble_lib but at this point, I would need filtering by MAC address (Background scan in Android return no results since 8.1 if no or empty filter is provided) and an option to disable location permission check (some phones allow ble scans without location permissions like many Honor and Huwaei).

Artur110, 2021-02-03:
@espositofulvio, thanks for your fast response :)
So you say that flutter_blue is now works in background? I'm trying to use it in onBackgroundMessage (notification handler method that called when app is on background or terminated) of firebase_messaging plugin.

But no luck here and I got error:
Unhandled Exception: MissingPluginException(No implementation found for method writeCharacteristic on channel plugins.pauldemarco.com/flutter_blue/methods)

Since you are using flutter_blue for similar task, maybe you faced this error?

espositofulvio, 2021-02-04:
@Artur110 I recall something like that happened before updating flutter_blue to the version implementing the new Flutter Plugin API. I can't really say what's happening.