PhilipsHue/flutter_reactive_ble #23: Can't scan with location service disabled

espositofulvio, 2020-02-22:
Hi,
I'm unable to scan for devices when the location services are disabled:

```D/ScanDevicesHandler$startDeviceScan$$inlined$let$lambda(26938): Error while scanning for devices:  Location Services disabled (code 4)```

I'm not sure what I'm doing wrong but I think I should be able to scan for devices when only the bluetoothservices are enabled.

I'm getting this on Android 8, with flutter_reactive_blue 1.1.0

I'm just doing something along the line of:

```dart
_scanSubscription = bleClient.scanForDevices(
  withService: BTDevice.serviceUuid,
  scanMode: ScanMode.lowLatency,
).listen(
  (device) {
  },
);
```

remonh87, 2020-02-23:
Thanks for reporting. By default location services is enabled and this is 
 because of https://issuetracker.google.com/issues/37065090 .  However little devices on Android7+ would require the location services to be enabled so I think I can create an option in the scansettings that disables it. For safety reasons I will keep it on by default

werediver, 2020-02-23:
@remonh87 Aren't the ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission is mandatory on Android to perform BLE scanning for privacy reasons (location information can be inferred from a BLE scan)? My understanding is there is nothing that can be done to work this around.

Am I wrong? What exactly are you planning to implement? (though, we can discuss this in a chat or off-line)

espositofulvio, 2020-02-23:
I'm not sure how they are doing it but I have other apps who can perform a scan without location services enabled, just the locaiton permission granted. Indeed, the same operation with FlutterBlue on the same phone worked with location services disabled.

remonh87, 2020-02-23:
@werediver  you are right the permissions are mandatory as defined by https://developer.android.com/guide/topics/connectivity/bluetooth-le#permissions. 
This actually does not mean the location service should be enabled (settings => location). The RxAndroidBle also supports the usecase where it is possible to allow scans in case location is turned off which is fine by Android. For more background info see: https://github.com/Polidea/RxAndroidBle/issues/106 and the issue logged in my previous comment. Let's discuss implementation details offline

PR #41 (2020-03-09): Support blescan with empty filter
Fixes #12 
Fixes #23
closed 2020-03-09 by remonh87