PhilipsHue/flutter_reactive_ble #27: Missing feature - Ability to Disconnect from a device

albinhubsch, 2020-02-24:
So I was digging in the code looking for a way to programmatically disconnect from a device and found the `DisconnectFromDeviceRequest`in the generated pb dart file. But it seems like this feature is not made available in the Method channel in any way? I would propose to add this as a feature making it possible to also disconnect from a device in the same way you're able to `_ble.connectToDevice(...`

werediver, 2020-02-24:
@albinhubsch When you connect to a device with `connectToDevice(⋯)` you get a stream delivering the connection status updates. Keep a subscription to this stream as long as you need the connection. By closing the subscription you terminate the conection.

werediver, 2020-02-24:
On the other hand, we _are_ missing documentation on this. Let's treat this ticket as a documentation request.

albinhubsch, 2020-02-25:
@werediver yes, documentation on this would be appreciated.

We've also tried what you proposed but without success. Saving the subscription and then cancel it does not seem to actually disconnect the the device connection.
```dart
_connectedDeviceSubscription = _ble.connectToDevice(id: widget.device.id, connectionTimeout: const Duration(seconds: 10)).listen(null);
... lifecycle ...
_connectedDeviceSubscription.cancel();
```

Our ble peripheral reports disconnection when the app closes/force closes, not when the subscription is canceled.

Taym95, 2020-02-25:
Hey, @albinhubsch we will add doc soon, take a look on this example:

https://github.com/Taym95/flutter_reactive_ble/blob/adbc381fcb115373de51fcb426596ff508f81a5e/example/lib/src/ui/device_detail_screen.dart#L53

If you call:

```
_currentValueUpdateSubscription.cancel();
``
You will disconnect.

werediver, 2020-02-25:
I'd indeed expect the disconnection mechanism to work fine, as @Taym95 confirms.

@albinhubsch Make sure you do not call `connectToDevice(⋯)` multiple times for the same device overriding your `_connectedDeviceSubscription` variable.

Also, be aware that the disconnection command on both iOS and Android is sort of a recommendation to the system stack, not an order. The BLE stack on both iOS and Android is designed to _share_ connections between applications, therefore the system is in its right to keep the connection open even after you close the connection on the app side. Can't say I observed that behavior a lot, but it is possible.

albinhubsch, 2020-02-25:
@werediver @Taym95 thanks for all the input! Will take it all into account doing some more testing with our devices.

remonh87, 2020-04-13:
Improved the docs usage should be more clear now

closed 2020-04-13 by remonh87

Ilya-Vys, 2023-05-15:
> I'd indeed expect the disconnection mechanism to work fine, as @Taym95 confirms.
>
> @albinhubsch Make sure you do not call `connectToDevice(⋯)` multiple times for the same device overriding your [...]

Is there an official docs, described that is sort of recommendation only?