PhilipsHue/flutter_reactive_ble #28: writeCharacteristicWithoutResponse fail

spd-heshuip, 2020-02-25:
writeCharacteristicWithoutResponse fail;

werediver, 2020-02-25:
@spd-heshuip Please care to provide enough input to enable us to help you resolve your issue.

Some points to check:

- make sure you use the correct service ID and characteristic ID that are actually present on the peripheral you're connected to
- BLE operations can fail, that's fine. Often a failure is followed by a connection loss, but there can be other reasons

spd-heshuip, 2020-02-25:
@werediver Hi，I have check my code again .  the serviceId and characteristic id is correct .and i use light blue to test ,it is work . but use the library it is fail;
[image]

Taym95, 2020-02-25:
What do you mean by `fail`? can you give the whole error output with reproducing example? only `writeCharacteristicWithoutResponse fail;` doesn't really help.

spd-heshuip, 2020-02-25:
here is my test code:
[image]
[image]
and here is the log:
[image]

spd-heshuip, 2020-02-25:
when i call the writeCharacteristicWithoutResponse function, it is block and did not return;

werediver, 2020-02-25:
@spd-heshuip I suspect your code does not use `flutter_reactive_ble` API correctly, because that's a pretty basic use case and I am sure it works. We use `writeCharacteristicWithoutResponse(⋯)` for peripherals' firmware update, for instance, and have no issues with that.

Please check your code agains the example app. If you still need assistance in resolving your issue, you'd have to provide a complete piece of code or an example project to reproduce the issue.

I see you're connecting to another smartphone in a peripheral role. Is it your app on the other end? Are you sure _that_ one is works correctly?

Are you able to perform write without response via another application line nRF Connect?

closed 2020-03-04 by werediver