PhilipsHue/flutter_reactive_ble #40: Example app scan doesn't work out of box - android

posipunk, 2020-03-04:
**Describe the bug**
Scanning returns no results when it should.

**To Reproduce**
Steps to reproduce the behavior:
1. Build example app. 
2. Run on Android (Galaxy Note 8, Android 9). 
3. After starting running it, give location permission (which is a different issue). 
4. Attempt to scan for a common UUID (i.e. 1800, 1801)

**Expected behavior**
Scan returns devices.

- [X ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Galaxy Note 8
 - OS: Android 9
 - Package version: Current master, March 2

**Peripheral device**
- Any! nRF Connect finds many devices around me.

**Additional context**
n/a

werediver, 2020-03-05:
> 3. After starting running it, give location permission (which is a different issue).

Please note, that this is _not_ an issue with this library, but a detail of Android privacy control mechanism. Permission to use Location Services on Android is required to perform BLE scanning, because BLE scanning can (indirectly) reveal information about your location.

Now, did you have Location Services _enabled_ when trying to perform scanning? It is currently required to have Location Services enabled to perform scanning, though this is indeed an issue (#23), which is going to be fixed in [v2.0.0](https://github.com/PhilipsHue/flutter_reactive_ble/milestone/1).

posipunk, 2020-03-05:
> > 1. After starting running it, give location permission (which is a different issue).
> 
> Please note, that this is _not_ an issue with this library, but a detail of Android privacy control mechanism. Permission to use Location Services on Android is required to perform BLE scanning, because BLE scanning can (indirectly) reveal information about your location.
> 
> Now, did you have Location Services _enabled_ when trying to perform scanning? It is currently required to have Location Services enabled to perform scanning, though this is indeed an issue (#23), which is going to be fixed in [v2.0.0](https://github.com/PhilipsHue/flutter_reactive_ble/milestone/1).

Regarding the location services - the example app starts up displaying "Authorize the FlutterReactiveBle example app to use Bluetooth and location" and then you need to leave the app and in app settings toggle the permission to get to the scanning screen. Not the ideal interface for an example app.

I did enable location services while scanning. Scanning returned nothing.

werediver, 2020-03-05:
@posipunk I'm afraid, there is no issues with the library. There is a big difference for a BLE service to be implemented by a peripheral (to be present in its GATT database) and to be advertised (to be included in a BLE advertisement packet). There is, for instance, "Device Information" primary service (see [GATT services](https://www.bluetooth.com/specifications/gatt/services/)) with a short ID 0x180A, which one would expect to be implemented on every device, and yet they would need a great deal of luck to find a device that advertises it.

posipunk, 2020-03-05:
> @posipunk I'm afraid, there is no issues with the library. There is a big difference for a BLE service to be implemented by a peripheral (to be present in its GATT database) and to be advertised (to be included in a BLE advertisement packet). There is, for instance, "Device Information" primary service (see [GATT services](https://www.bluetooth.com/specifications/gatt/services/)) with a short ID 0x180A, which one would expect to be implemented on every device, and yet they would need a great deal of luck to find a device that advertises it.
> 
> Please, take a generic BLE scanner app like nRF Connect and look for BLE peripherals in your vicinity that advertise at least some services, then use the IDs of those services in the example app.
> 
> Things will become less confusing for newcomers, when #12 will be implemented, because then it will be possible to scan for devices specifying no services at all (coming in [v2.0.0](https://github.com/PhilipsHue/flutter_reactive_ble/milestone/1)).

Thank you for your comment - think I've solved the mystery.
I've been using nRF Connect for most of my debugging but realized it also tells which services are in the advertisement packet.
Appears the scan only looks for advertised services - it will not find devices based on services that have to be found via discovery.
Interestingly my device happens to advertise 0x180A and I was able to find it when I scanned for that one.
Would be good to clarify this in the documentation.

closed 2020-03-05 by posipunk