PhilipsHue/flutter_reactive_ble #43: No way of knowing if peripheral is connectable

juliansteenbakker, 2020-03-11:
**Is your feature request related to a problem? (please describe)**
Right now it's not possible to check if the found peripheral is connectable. There are devices that advertise without being connectable, so it would be good to filter out those and only allow connectable devices.

**Describe the solution you'd like**
For iOS, CBAdvertisementDataIsConnectable lets you know wether a peripheral is connectable. It would be great if this gets passed through.
However, since this plugin relies on RxAndroidBle, it is currently not possible to check if a peripheral is connectable on Android. See ticket https://github.com/Polidea/RxAndroidBle/issues/668

**Describe alternatives you've considered**
x

**Additional context**
Add any other context or screenshots about the feature request here.

remonh87, 2020-03-11:
Thanks for reporting this I think this will be a good addition.

RtgrV, 2023-06-28:
@Taym95 we should be able to close this issue after releasing a new version.

closed 2023-08-29 by Taym95