PhilipsHue/flutter_reactive_ble #44: Read characteristics on iOS is not working

bulenthacioglu, 2020-03-12:
**Describe the bug**
When a device connection is established, characteristics cannot be read on iOS, whereas on Android it works without any problem. Following error is printed:

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectTo()` 
2. Subscribe to a non-protected characteristic
3. Observe a failure with following exception 

> Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at $103218a20).Failure:1, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $103218a20).Failure error 1.), {})

```_connectedDeviceStream = _ble.connectToDevice(id: deviceId, connectionTimeout: const Duration(seconds: 20)).asBroadcastStream();
    _connectedDeviceStream.where((device) => device.connectionState == DeviceConnectionState.connected).listen((onData) async {
      var chars = _ble.subscribeToCharacteristic(
        QualifiedCharacteristic(characteristicId: _charUuid, serviceId: _serviceUuid, deviceId: deviceId),
      );
      var listOfChars = await chars.toList();
      print('chars ${listOfChars.runtimeType} $listOfChars');
      _values.add(listOfChars);
    });
```

**Tablet**
 - Device: iPad (7th generation)
 - OS: iOS 13.3.1

werediver, 2020-03-12:
@bulenthacioglu First of all, you are mixing reading and subscribing up. Reading is a one time operation. Subscribing is a continuous listening to notifications or indications from a characteristic.

Second, `_ble.subscribeToCharacteristic(⋯)` returns a stream of data packets, not just data. You are treating it as just data.

bulenthacioglu, 2020-03-12:
Sorry, you are right. I modified my code accordingly but still get the same error.

```_connectedDeviceStream = _ble.connectToDevice(id: deviceId, connectionTimeout: const Duration(seconds: 20)).asBroadcastStream();
    _connectedDeviceStream.where((device) => device.connectionState == DeviceConnectionState.connected).listen((onData) async {
      var chars = QualifiedCharacteristic(characteristicId: _charUuid, serviceId: _serviceUuid, deviceId: deviceId);
      _ble.subscribeToCharacteristic(chars).listen((data){
        print('data is $data');
        _values.add(data);
      });
    });

werediver, 2020-03-12:
Can you show how `_connectedDeviceStream` is managed?

bulenthacioglu, 2020-03-12:
`    _connectedDeviceStream = _ble.connectToDevice(id: deviceId, connectionTimeout: const Duration(seconds: 20)).asBroadcastStream();
`

werediver, 2020-03-12:
Wait a sec. The error says: "Error **unsubscribing** from notifications". It happens when you loose the connection, and in that case it's sort of normal (expected).

werediver, 2020-03-12:
I hope you're not doing any of this in a widget's `build()` method?

bulenthacioglu, 2020-03-12:
Yeah, that I am aware of but why would I lose the connection when I am trying to subscribe the characteristics?

bulenthacioglu, 2020-03-12:
I am doing these in `initState()`

werediver, 2020-03-12:
It's (most likely) not related, the connection loss and subscribing to a characteristic.

Do you get any data before the error?

werediver, 2020-03-12:
And what are you connecting to? You didn't properly fill in the issue template, unfortunately

bulenthacioglu, 2020-03-12:
In my initial code, I was getting an empty list, on Android it used to get the data without any problem.

Second version, it crashes before getting the data, Android works still.

I am trying to connect to Beurer BM54 Blood pressure monitor

werediver, 2020-03-12:
A crash means the app disappears from the screen, in which case you can get a stack-trace from the native code. Can you provide the crash log?

bulenthacioglu, 2020-03-12:
It is not crashing, it throws the exception. My bad in explanation. There is no stack trace and as I told it is the following:

> Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at $103218a20).Failure:1, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $103218a20).Failure error 1.), {})

werediver, 2020-03-12:
That error by itself is nothing special, because likely it means the connection has been lost, that's it.

I'd look for why the connection can be lost. Is it a known bug in your peripheral? Are you unsubscribing from the stream, returned from `_ble.connectToDevice()`? (because that's the way to disconnect)

werediver, 2020-03-12:
Also: does the official app work with the sensor from the device on which your implementation fails?

bulenthacioglu, 2020-03-12:
Yes it works. Also, as I told the library works with Android device (Pixel 3).

werediver, 2020-03-12:
I'm sorry, but this:

> In my initial code, I was getting an empty list, on Android it used to get the data without any problem.

Does not sound realistic, because in your initial code you were treating a stream object as data.

Perhaps, if you provide a more complete piece of code, we can see something. Otherwise I recommend to focus on keeping the connection alive and investigating why you can loose it.

bulenthacioglu, 2020-03-13:
I changed my code to reading the characteristic rather than subscribing. Now I have a different exception.

```
_ble.connectToDevice(
      id: deviceId,
      servicesWithCharacteristicsToDiscover: {
        _serviceUuid: [_charUuid]
      },
      connectionTimeout: const Duration(seconds: 2),
    ).listen((connectionState) async {
        final characteristic = QualifiedCharacteristic(serviceId: _serviceUuid, characteristicId: _charUuid, deviceId: deviceId);
        List<int> response = await _ble.readCharacteristic(characteristic);
        print('response is $response');
      print('connection state is $connectionState');
    }, onError: (Object error) {
      print('connection error is $error');
    });
```
```
#0      Result.dematerialize.<anonymous closure> (package:flutter_reactive_ble/src/model/result.dart:18:13)
#1      Result.iif (package:flutter_reactive_ble/src/model/result.dart:29:21)
#2      Result.dematerialize (package:flutter_reactive_ble/src/model/result.dart:11:28)
#3      FlutterReactiveBle.readCharacteristic.<anonymous closure> (package:flutter_reactive_ble/src/reactive_ble.dart:118:40)
#4      _MapStream._handleData (dart:async/stream_pipe.dart:229:21)
```

werediver, 2020-03-13:
You've shown the stack trace, but not the exceptions description. Can you provide that as well?

werediver, 2020-03-13:
Now you are trying to read on every connection status update, not only when the device is gets connected.

bulenthacioglu, 2020-03-13:
Unfortunately there is no description on the stack trace.

bulenthacioglu, 2020-03-13:
For the version 1.1.0, the example app also cannot read the characteristics btw. In the new version(2.0.0), there is no example for reading anyways if I am not mistaken.

werediver, 2020-03-13:
Right, the example app has been refactored to demonstrate a better overall architecture and only supports scanning and connecting at the moment.

@bulenthacioglu You keep making very basic mistakes in your code. Please, spend time on thoroughly checking your program and thinking through the possible scenarios around what you're doing. BLE programming is not simple: it requires some understanding of the protocol as well as the BLE stack peculiarities on both iOS and Android, as well as understanding peculiarities of your particular BLE peripheral. The ability to write correct Dart code and properly managing all the asynchronous operations and any related resources (e.g. subscriptions) is an absolute must. Understanding the lifecycle of Flutter UI is an absolute must, if UI is involved in your work.

I perfectly understand that lack of documentation and a limited example app doesn't help you to get through. We are working on it directly as well as accepting contributions. We expect to significantly improve on both points _with time_.

But sometimes it's necessary to learn a new complex topic, iteratively build up and debug tricky asynchronous code, and dig deeper into an underdocumented API or even its implementation to get things right.

bulenthacioglu, 2020-03-13:
Ok, apparently this is some issue we cannot resolve here. I will be waiting for a more proper documentation from you for the iOS part and at the meantime will try to discover the library on my own. 
If you are ever interested in the possible shortcomings of the library, I would suggest checking out an old version (i.e. 170d3fb) and try the same code on iOS and Android. I am not saying the library is problematic, maybe my understanding is limited (which I agree on Bluetooth side). 

I only changed the variables `_serviceUuid` and `_charUuid` according to the device I used. 
On Android I get the error :

> [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "GATT exception from MAC address 3C:42:7E:05:8C:A1, with type BleGattOperation{description='CHARACTERISTIC_READ'}")

but nevertheless it **succeeds** on reading the characteristics. On iOS, it cannot read any characteristic and throws the following exception:

> Unhandled Exception: Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "The characteristic 0000... of the service 0000... of the peripheral 39D... is not readable")

Again, this is from the example provided by the library, not my code. Maybe the device I am working with is problematic, I don't know that either. Anyways, I hope you provide a more thorough documentation soon.

All the best :)

remonh87, 2020-03-17:
We will spend coming times in improving the docs and hopefully make it more clear how to use this library. For now I will close this ticket.

closed 2020-03-17 by remonh87

ch-muhammad-adil, 2020-07-30:
So dear I am facing same issue, It works fine in android but for iOS it is not subscribing, It takes time and sends back this error. 

`Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at $10756bed0).Failure:1, The operation couldn't be completed. (flutter_reactive_ble.Central.(unknown context at $10756bed0).Failure error 1.), {})`

Here is the subscription method 
````
           currentDeviceConnector.bleObject
                                      .subscribeToCharacteristic(
                                    QualifiedCharacteristic(
                                      deviceId: oxygenDevice.id,
                                      serviceId: Uuid.parse(
                                          "F000EFE0-0000-4000-0000-00000000B000"),
                                      characteristicId: Uuid.parse(
                                          "F000EFE3-0451-4000-0000-00000000B000"),
                                    ),
                                  )
                                      .listen((data) async {
                                    print("Data from oximeter device: " +
                                        data.toString());
                                  });
````

Here are some more logs & screenshot from xCode 

````
      Assertion failed: file /Users/user/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.3.0/ios/Classes/Plugin/PluginController.swift, line 99
2020-07-30 14:27:04.177988-0400 Runner[2228:2367463] Assertion failed: file /Users/user/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.3.0/ios/Classes/Plugin/PluginController.swift, line 99

`````

[image]

ch-muhammad-adil, 2020-07-30:
I really need this to work because we are almost there to launch this product in next 2-3 days and this is going to be main issue which will block us from launching.

werediver, 2020-07-31:
@ch-muhammad-adil Please care to open a new ticket and provide as much context for you issue as possible. Take into account that the original issue likely involved some implementation mistakes. It may be the case that you have similar implementation mistakes, but that will remain unknown, unless the implementation is shared (by you) and analyzed (by us).

In the meantime we can assure you that the library is operational on both platforms, which is demonstrated by [Philips Hue Bluetooth](https://www.philips-hue.com/en-us/app/bluetooth) app.

ch-muhammad-adil, 2020-07-31:
Hello Again, I have provided very detailed information and this plugin has issue for iOS. Just try BottomSheet for displaying devices and connecting to these devices. I am returning BLE connector object back to my main view also the connected device in a callback. There i am trying to subscribe, it is working very well in android but for iOS subscription won't happen and what i have shared I think should be enough information.

werediver, 2020-07-31:
@ch-muhammad-adil I struggle to imagine how a platform exception and a platform level assertion can happen at the same time, which suggests you may be talking about two separate failures.

It is important what triggers your code for establishing a subscription (a button click? an init-state method? a build-method?). The same has been indicated to the author of the original issue.

I still suggest to handle your issue in a separate ticket, where you can fill in the proposed issue-template and provide any other relevant information, as well as a link to any other tickets you find relevant.

After all, if you do not receive enough help from us, you have to help yourself. If your investigation uncovers an issue in the library for which you can provide a fix, please open a pull-request.

farr64, 2020-07-31:
@werediver: On a similar note, you might want to mention to your Philips Marketing Department that, motivated by your technical comment, I purchased a Philips Hue White Ambiance A19 LED Smart Bulb and I downloaded your Philips Hue Bluetooth app.

It's all for science and engineering but, hey, it's also a small (and meaningful) way to thank Philips for its wonderful action of sharing your software technology via Open Source. 

In the end, all of us benefit and "all" includes Philips.