PhilipsHue/flutter_reactive_ble #48: Using device.name as a discriminant to select (or ignore) discoverable devices

farr64, 2020-03-27:
It would be useful to have the ability to use **device.name** (or a portion, perhaps the first couple of characters) as a discriminant to select (or ignore) discoverable devices.

I explored your use of `_isValidUuidInput` in `device_list.dart`  but I could not quite come up with an optimum alternative .

Could please help with an example along the lines of the following, which is obviously incorrect in Dart?

`device.name.startsWith('xyz')`

Thanks.

Alfredo

remonh87, 2020-03-27:
Hi you mean something like: 

```dart
 _ble.scanForDevices(withServices: serviceIds)
        .where((bleDevice)=> bleDevice.name.contains("foo"))
```

farr64, 2020-03-27:
Precisely. Thanks very much.

I just started learning about Flutter and I'm very interested in learning all I can.

Your package looks (and feels) solid.

 I'm sure that, once I get the hang of things, this will be a very enjoyable journey.

Hartelijk dank.

remonh87, 2020-03-27:
You are welcome feel free to to contact us in case you have other questions

closed 2020-03-27 by remonh87