PhilipsHue/flutter_reactive_ble #51: Error while scanning for devices: Location Permission missing

mbn18, 2020-04-05:
**Describe the bug**
Get the following error when trying to scan:

> E/flutter: [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Location Permission missing (code 3)")

**To Reproduce**
Steps to reproduce the behavior:
1. Installed flutter_reactive_ble: ^2.0.0
2. Implemented  [bleScanner](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/src/ble/ble_scanner.dart)
3. scannerObj.startScan();

**Expected behaviour**
Request for `ACCESS_FINE_LOCATION` permission

**Smartphone / tablet**
 - Device: Samsung A5 2017
 - OS: Android 8.0.0
 - Package version: 2.0.0

**Peripheral device**
- Vendor, model: raspberry pi
- Does it run a custom firmware / software: [go-ble](https://github.com/go-ble/ble)

**Additional context**
Setup is working with [flutter_blue](https://pub.dev/packages/flutter_blue).

remonh87, 2020-04-06:
Requesting permissions is something that should be handled in the application in my opinion. One can choose for an automated permissionchecker and some else uses an explanation screen. 

However if you use:

```dart
 _ble.scanForDevices(withServices: serviceIds, requireLocationServicesEnabled =false) 
```

you ignore the location services should be enabled flag.  We have this flag enabled by default since it will yield flaky behavior on Pixel and Nexus devices

mbn18, 2020-04-07:
If I understand correctly, the flag `requireLocationServicesEnabled` do not trigger permission request. It seems to check if the service is enabled or not.

And that mean we need to use external logic to auth `ACCESS_FINE_LOCATION` permission.

Just to be sure, I did checked both variant (true/false) and none triggered the permission auth message.

remonh87, 2020-04-07:
you can use the `_ble.statusStream` to listen to the current ble status. In case the permission is not there we will yield `unauthorized`.  It is then up to the application that uses our library to request the permissions programatically.

mbn18, 2020-04-07:
Great.
If I may suggest, it would be helpful to add a permission section to the `README.md` to explain what permission are needed for each task/platform.

PR #53 (2020-04-18): add ble status usage to readme
Create a small pr to add BLE status to readme since it is core functionality and it was not clear we support this main use case.

Fixes #51
closed 2020-04-18 by remonh87