PhilipsHue/flutter_reactive_ble #52: minSdk API lvl 24 as a lib requirement

vsxed, 2020-04-10:
**Describe the bug**
The libs minSdk requirement for Android should be lowered, since BLE has been introduced in API level 18/19, but the library is requiring API level 24

**To Reproduce**
Try to build the example app with minSdk set to API lvl 19

**Expected behavior**
Should be building while i drink my coffee

- [x] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Nexus 4, 4.4 Kitkat

**Peripheral device**
- 

**Additional context**
-

remonh87, 2020-04-11:
Thanks for reporting the issue. When we started developing the library we decided to first make this library work well on Android 7 and higher. The main reason for this choice is the fragmentation on Android and the big difference in BLE behavior between Android 6 and 7+ . However I am willing to lower the min sdk if more people request this feature or are willing to participate in testing our library on lower Android versions.

vsxed, 2020-04-11:
Thank you very much for your consideration!

edsonboldrini, 2020-07-30:
I'd like to contribute testing the lib in a android api 19, my clients are needing this.

remonh87, 2020-07-30:
@edsonboldrini  thanks for helping us out. I am willing to lower the sdk but api 19 is quite low I see some issues:

- Our library is written in Kotlin and I am not sure if it will run on Android4.0. I somewhere read that Kotlin can crash on Android versions below 5.
- Bluetooth LE has evolved a lot and older Android phones will have outdated chipsets so be aware which features you use (e.g. Link layer encryption is only supported since BLE version 4.2 )

I propose to create a branch where I lower the Android sdk version to 19 and feel free to test it. Will create it tomorrow. Let's  create separate issues for things we encounter. Sounds like a plan?

edsonboldrini, 2020-07-30:
@remonh87 Looks a great plan. In my PC I've already downgrade the lib and the example project just for test. If you prefer, I could fork the project and create a branch doing the downgrades, after some tests a could make some pull requests for you. Until now no problems testing on Android 4.4.4 SDK 19.

[image]

remonh87, 2020-07-30:
That is even easier for me 👍

edsonboldrini, 2020-07-30:
Ok, I'll wait for you! Tag me into this issue when you're done. 
Best regards!

remonh87, 2020-07-31:
@edsonboldrini  I meant to say your proposed idea of using your fork for testing is perfect. If you see improvements feel free to create a pr for it, also reach out if you have some questions.

edsonboldrini, 2020-07-31:
I'm sorry about this misunderstanding, I'll fork today!

Arley011, 2020-10-16:
> However I am willing to lower the min sdk if more people request this feature or are willing to participate in testing our library on lower Android versions.

Would be really nice :)

ened, 2020-11-07:
@remonh87 what kind of testing report would you like to see in order to gain confidence for making this library minSdk21 by default? We have a test wall with a lot of Android phones of all API levels and can do structured testing.

I also understand that you are relying on RxAndroidBle, so IMHO the decision on minSdk should really be upstream - if RxAndroidBle supports it, so should this library.

remonh87, 2020-11-08:
@ened thank you for message. For me it not needed to gain fully confidence it works, I am quite confident myself it can given the good quality of the underlying library of Polidea and the effort we put in it ourselves. The problem is more that we need help from the community in case bugs happen on Android version <7.  since it becomes more challenging to triage them ourselves and test the fix. If you and others are willing to help out in testing the library I am more then happy to lower the sdk on Android (with above disclaimer).

ened, 2020-11-08:
@remonh87 happy to help - again, we have hardware available to run some tests.

If only RxAndroidBle had a verification test suite we could run to test things automatically. 🤔 

> Also keep in mind (probably you know already) that when you develop for Android 5 not all devices support BLE so you need to make sure you make the app only available to devices that support that feature.

I'd go with this: https://developer.android.com/guide/topics/connectivity/bluetooth-le and set the following in **AndroidManifest.xml** for those devices:

```
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
```

Understand users may still install via 3rd party or APK which steps around the `uses-feature` but 99% of users do not.

closed 2020-11-13 by remonh87

remonh87, 2020-11-13:
Lowered the minsdk to 21. This fix is merged into master and will be part of version 2.7.0 which will be released today

ened, 2020-11-13:
@remonh87 thank you so much for this.

PR #141 (2020-11-13): Lower minsdk android
Fixes #52