PhilipsHue/flutter_reactive_ble #55: Error compiling for iOS fatal error: 'flutter_reactive_ble/flutter_reactive_ble-Swift.h' file not found

marcel-bluestone, 2020-04-21:
When compiling v2.0.0 for iOS device, I receive the fatal error: 'flutter_reactive_ble/flutter_reactive_ble-Swift.h' file not found

I can't find the file in the package. Can you help me?

flutter doctor output:
flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel master, v1.18.0-6.0.pre.115, on Mac OS X 10.15.4 19E287, locale de-DE)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 11.4.1)
[✓] Android Studio (version 3.5)
[✓] Connected device (1 available)

• No issues found!

werediver, 2020-04-22:
The "-Swift.h" files are autogenerated. Looks like an Xcode (or some other) glitch. I'd suggest to clean up your working copy and Xcode related caches, etc., and also to check this: [Flutter: Xcode 11.4 Support](https://flutter.dev/docs/development/ios-project-migration).

To clean up your working copy consider using `flutter clean` and/or `git clean` (perhaps, `git clean -dfx`, but **read the manual first** to avoid data loss).

To clean (some of) the Xcode related caches consider using the following commands:

```sh
xcrun -k
rm -rf ~/Library/Caches/com.apple.dt.Xcode
rm -rf ~/Library/Developer/Xcode/DerivedData
rm -rf "$(getconf DARWIN_USER_CACHE_DIR)/org.llvm.clang/ModuleCache"
rm -rf "$(getconf DARWIN_USER_CACHE_DIR)/org.llvm.clang.$(whoami)/ModuleCache"
```

Also, try building using both `flutter build ios --debug` (or from your IDE) and from Xcode, it might give different results/insights.

remonh87, 2020-04-22:
From the flutter doctor output I noticed you are using a very bleeding edge flutter version. We just released v2.1.0 of the app that  adds support for the new plugin structure. I tested our library with the latest beta version today and everything compiles at my machine. 

If v2.1.0 does not solve your issue I would suggest to try the steps  @werediver mentioned

marcel-bluestone, 2020-04-22:
OK, so I followed werediver's steps to clear the caches, upgraded the package to 2.1.0 and tried to build using flutter build ios --debug... Did not help unfortunately:

```
flutter build ios --debug
 
Building com.bluestone-tec.sensorMapping for device (ios)...
Automatically signing iOS for device deployment using specified development team in Xcode project: VCY32VM6H4
Running Xcode build...                                                  
                                                   
Xcode build done.                                           18,9s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    111 warnings generated.
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:546:49: warning:
    'CBManagerState' is only available on iOS 10.0 or newer [-Wunguarded-availability]
    - (ProtosBluetoothState*)toBluetoothStateProto:(CBManagerState)state {
                                                    ^
    In module 'CoreBluetooth' imported from
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/System/Library/F
    rameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced
    in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:546:1: note:
    annotate 'toBluetoothStateProto:' with an availability attribute to silence this warning
    - (ProtosBluetoothState*)toBluetoothStateProto:(CBManagerState)state {
    ^
                                                                          API_AVAILABLE(ios(10.0))
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:74:37: warning:
    'CBManagerStateUnsupported' is only available on iOS 10.0 or newer [-Wunguarded-availability]
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/System/Library/F
    rameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced
    in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:74:37: note:
    enclose 'CBManagerStateUnsupported' in an @available check to silence this warning
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:74:95: warning:
    'CBManagerStateUnknown' is only available on iOS 10.0 or newer [-Wunguarded-availability]
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                                                                                  ^~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/System/Library/F
    rameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note: 'CBManagerState' has been marked as being introduced
    in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:74:95: note:
    enclose 'CBManagerStateUnknown' in an @available check to silence this warning
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                                                                                  ^~~~~~~~~~~~~~~~~~~~~
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.m:80:37: warning:
    'CBManagerStatePoweredOn' is only available on iOS 10.0 or newer [-Wunguarded-availability]
        if(self.centralManager.state == CBManagerStatePoweredOn) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.1+1/ios/Classes/FlutterBluePlugin.h:6:
[... 21 lines cut ...]
    extras/BLEStatus.swift:4:5: note: handle unknown values using "@unknown default"
        switch (centralState) {
        ^
[... cut ...]
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.1.0/ios/Classes/ReactiveBlePlugin.m:2:9: fatal
    error: 'flutter_reactive_ble/flutter_reactive_ble-Swift.h' file not found
    #import <flutter_reactive_ble/flutter_reactive_ble-Swift.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description

Encountered error while building for device.

```

werediver, 2020-04-22:
@marcel-bluestone First, this your log indicates an issue different from what you mentioned initially, so, I guess, the initial issue has been solved by some of the cleaning steps.

Second, the log is quite informative. Have you analyzed it? Pay attention to this, in particular:

```
'CBManagerState' has been marked as being introduced
    in iOS 10.0 here, but the deployment target is iOS 8.0.0
```

Is your deployment target set to iOS 8.0? Unfortunately, iOS 10.0 is required for the current implementation to work. If iOS 8.0 as the deployment target is not a hard requirement for you, just change it to iOS 10.0.

marcel-bluestone, 2020-04-22:
Thanks for your update. The problem is that I am using multiple packages, and each one seems to automatically set a different deployment target for the package. So the one at the project level says 12.4. But I did a search and replace now, and set all of the deployment targets in the project to 12.4.

However the error is still there:

`flutter build ios --debug
Building com.bluestone-tec.sensorMapping for device (ios)...
Automatically signing iOS for device deployment using specified development team in Xcode project: VCY32VM6H4
Running Xcode build...                                                  
                                                   
Xcode build done.                                           15,2s
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:189417:26: warning: ambiguous expansion of macro 'MIN' [-Wambiguous-macro]
          p1->aCoord[ii].f = MIN(p1->aCoord[ii].f, p2->aCoord[ii].f);
                             ^
    In module 'Darwin' imported from /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:1077:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/usr/include/sys/param.h:215:9: note: expanding this definition of 'MIN'
    #define MIN(a, b) (((a)<(b))?(a):(b))
            ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:13916:10: note: other definition of 'MIN'
    # define MIN(A,B) ((A)<(B)?(A):(B))
             ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:189418:28: warning: ambiguous expansion of macro 'MAX' [-Wambiguous-macro]
          p1->aCoord[ii+1].f = MAX(p1->aCoord[ii+1].f, p2->aCoord[ii+1].f);
                               ^
    In module 'Darwin' imported from /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:1077:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/usr/include/sys/param.h:218:9: note: expanding this definition of 'MAX'
    #define MAX(a, b) (((a)>(b))?(a):(b))
            ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:13919:10: note: other definition of 'MAX'
    # define MAX(A,B) ((A)>(B)?(A):(B))
             ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:189423:26: warning: ambiguous expansion of macro 'MIN' [-Wambiguous-macro]
          p1->aCoord[ii].i = MIN(p1->aCoord[ii].i, p2->aCoord[ii].i);
                             ^
    In module 'Darwin' imported from /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:1077:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/usr/include/sys/param.h:215:9: note: expanding this definition of 'MIN'
    #define MIN(a, b) (((a)<(b))?(a):(b))
            ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:13916:10: note: other definition of 'MIN'
    # define MIN(A,B) ((A)<(B)?(A):(B))
             ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:189424:28: warning: ambiguous expansion of macro 'MAX' [-Wambiguous-macro]
          p1->aCoord[ii+1].i = MAX(p1->aCoord[ii+1].i, p2->aCoord[ii+1].i);
                               ^
    In module 'Darwin' imported from /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:1077:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/usr/include/sys/param.h:218:9: note: expanding this definition of 'MAX'
    #define MAX(a, b) (((a)>(b))?(a):(b))
            ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:13919:10: note: other definition of 'MAX'
    # define MAX(A,B) ((A)>(B)?(A):(B))
             ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:189474:12: warning: ambiguous expansion of macro 'MAX' [-Wambiguous-macro]
          x1 = MAX(DCOORD(p->aCoord[jj]), DCOORD(aCell[ii].aCoord[jj]));
               ^
    In module 'Darwin' imported from /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:1077:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.4.sdk/usr/include/sys/param.h:218:9: note: expanding this definition of 'MAX'
    #define MAX(a, b) (((a)>(b))?(a):(b))
            ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:13919:10: note: other definition of 'MAX'
    # define MAX(A,B) ((A)>(B)?(A):(B))
             ^
    /Users/marcel/StudioProjects/sensor_mapping/ios/Pods/SQLCipher/sqlite3.c:189475:12: warning: ambiguous expansion of macro 'MIN' [-Wambiguous-macro]
          x2 = MIN(DCOORD(p->aCoord[jj+1]), DCOORD(aCell[ii].aCoord[jj+1]));
               ^
[... 176 lines cut ...]
    values, possibly added in future versions
        switch (centralState) {
        ^
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.1.0/ios/Classes/BleData extras/BLEStatus.swift:4:5: note: handle unknown values using "@unknown default"
        switch (centralState) {
        ^
[... cut ...]
        switch connectionState {
        ^
    /Users/marcel/.pub-cache/hosted/pub.dartlang.org/sqflite_sqlcipher-1.0.0+6/ios/Classes/SqfliteSqlCipherPlugin.m:654:15: warning: treating Unicode character as whitespace [-Wunicode-whitespace]
            } else {
                  ^
    1 warning generated.
    1 error generated.
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description

Encountered error while building for device.
`

werediver, 2020-04-22:
@marcel-bluestone "The error" vanished after you cleaned up Xcode caches. Then there was a deployment target unsupported by "flutter_reactive_ble", which you "fixed" (I'm afraid, not in the right way). This is yet another issue (and the log does not contain the actual error, only notes and warnings).

Regarding the deployment target, please set at least iOS 10.0 in _your app_ build settings, not anywhere else. You should not modify anything managed by CocoaPods.

Your new issue does not appear to be logged in the excerpt you included, therefore I cannot give you a hint this time. But I'd recommend you to research such general issues on your own, because they are not specific to this library (or ask questions on Stack Overflow, a relevant subreddit, or your favorite forum).

remonh87, 2020-04-28:
For now I close the issue. If you still encounter problems you can reopen it or log another issue.

closed 2020-04-28 by remonh87