PhilipsHue/flutter_reactive_ble #58: Connecting with connectionTimeout parameter fails immediately with GATT ERROR 133

vsxed, 2020-04-24:
**Describe the bug**
Connecting to a device with a connectionTimeout value given, results most of the times in a GATT ERROR 133 (see the logcat log attached below). Not having a connectionTimeout parameter set results in a successful (and pretty fast) connection.

Seems like an android-only issue, the same code works with iOS just fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `.connectToDevice(id: ..., connectionTimeout: Duration(seconds: 10))`
2. See the failed attempt to connect in you log

**Expected behavior**
The expected behavior would be an established connection or a timeout after the set duration

- [ ] I tried doing the same with a general BLE scanner application

**Smartphone / tablet**
 - Device: OnePlus 5t
 - OS: Android 9
 - Package version: 2.0.0

**Peripheral device**
- Vendor, model: Custom
- Does it run a custom firmware / software: no

**Additional context**
```
E/AccessibilityBridge( 7512): VirtualView node must not be the root node.
D/PluginController( 7512): Start connecting for device 78:DB:2F:D0:BC:D2
D/PluginController( 7512): Start connecting for device 78:DB:2F:D0:BC:D2
D/DeviceConnectionHandler( 7512): Try to connect to device 78:DB:2F:D0:BC:D2
D/DeviceConnectionHandler( 7512): Try to connect to device 78:DB:2F:D0:BC:D2
D/DeviceConnector( 7512): Connecting device 78:DB:2F:D0:BC:D2
D/DeviceConnector( 7512): Connecting device 78:DB:2F:D0:BC:D2
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=0
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=0
D/BluetoothGatt( 7512): connect() - device: 78:DB:2F:D0:BC:D2, auto: false
D/BluetoothGatt( 7512): registerApp()
D/BluetoothGatt( 7512): registerApp() - UUID=d1024742-f90f-4f43-ab6d-1f9fbfe29569
D/BluetoothGatt( 7512): onClientRegistered() - status=0 clientIf=9
I/flutter ( 7512): ### CONNECTING ###
E/AccessibilityBridge( 7512): VirtualView node must not be the root node.
E/AccessibilityBridge( 7512): VirtualView node must not be the root node.
D/BluetoothGatt( 7512): onClientConnectionState() - status=133 clientIf=9 device=78:DB:2F:D0:BC:D2
D/DeviceConnector$establishConnection( 7512): Connection established for device 78:DB:2F:D0:BC:D2
D/DeviceConnector$establishConnection( 7512): Connection established for device 78:DB:2F:D0:BC:D2
D/ReactiveBleClient$connectToDevice( 7512): Connect 78:DB:2F:D0:BC:D2 fails: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 133 (GATT_ERROR)
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=0
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=0
D/ReactiveBleClient$connectToDevice( 7512): Connect 78:DB:2F:D0:BC:D2 fails: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 133 (GATT_ERROR)
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Error conn device:  78:DB:2F:D0:BC:D2 error=Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 133 (GATT_ERROR)
I/chatty  ( 7512): uid=10560(com.camparound.truma.cooler.truma_cooler) identical 2 lines
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Error conn device:  78:DB:2F:D0:BC:D2 error=Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 133 (GATT_ERROR)
D/BluetoothManager( 7512): getConnectionState()
D/BluetoothManager( 7512): getConnectedDevices
I/flutter ( 7512): ### CONNECTING ###
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=2
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=2
D/BluetoothGatt( 7512): close()
D/BluetoothGatt( 7512): unregisterApp() - mClientIf=9
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=3
D/DeviceConnectionHandler$listenToConnectionChanges( 7512): Update device: 78:DB:2F:D0:BC:D2 status=3
D/PluginController( 7512): stop notifications
I/chatty  ( 7512): uid=10560(com.camparound.truma.cooler.truma_cooler) identical 12 lines
D/PluginController( 7512): stop notifications
I/flutter ( 7512): ### CONNECTION FAILED ###: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 133 (GATT_ERROR)") ~~ with DeviceConnectionState.disconnected
D/PluginController( 7512): Disconnect device: 78:DB:2F:D0:BC:D2
D/PluginController( 7512): Disconnect device: 78:DB:2F:D0:BC:D2
E/AccessibilityBridge( 7512): VirtualView node must not be the root node.
```

remonh87, 2020-04-28:
Thanks for reporting. There is a fundamental difference in connecting on Android when there is a timeout specified and where there is not. In case a timeout is specified we will connect without autoconnect (since we want to connect quick and timeout after certain time). In case there is **no** timeout specified we use the autoconnect is true flag which asks Android to connect when the device becomes available. The last flag yields to more stable connection behavior but for some phones also in much slower connection speed ( we did some benchmarking when developing our BLE app and decided to go for autoconnect = false). More info about autoconnect see https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback) 

The GATT 133 error is a very generic error and in general it means the Phone and the BLE peripheral are not properly synchronised.  Question: do you directly connect after scanning? If so can you maybe wait for a second and then try to connect?

vsxed, 2020-04-28:
Hi @remonh87, thanks for your answer so far!

The only thing i tried was to set the timeOut. Further logic is handled by a specific bloc, which works with your lib. But I do understand the decision you did with setting `autoconnect=false` when using a timeout.

> Question: do you directly connect after scanning? If so can you maybe wait for a second and then try to connect?

Yeah, pretty immediately. I do `await` the cancellation of the discovery StreamSubscription and then try to connect to the device:

```
void handleBleConnection(BleConnectEvent event) async {
    await cancelDiscoveryScan();
    await cancelCurrentDeviceConnection();

    ...

    _connectToTargetDevice = _instance.connectToDevice(id: event.target.id).listen((update) {
      this.add(BleConnectionStateEvent(event.target, update));
    });
}
```

Setting a fixed delay before connecting to the peripheral seems somehow odd to me, since devices do behave differently in this case - some connect immediately (like iOS devices), some may need a delay of 1s, some may need a short or longer delay. Feels like some kind of trial and error.

Would it be possible to add an `autoconnect` option, next to `timeout` to gather more control about how the connection will be established? Could still be defaulting to `false`.

Nevertheless, as of now i did a workaround (my own timeout implementation), which works nice, but adds some (unnecessary) boilerplate to the codebase. But i will try to set some delays between the discovery scan and the connection.

werediver, 2020-04-28:
Be aware that _your_ timeout is likely to let some resources slip through in more cases than the timeout implemented at a lower level in the plug-in. Not sure about the details at the moment, but I can imagine that just unsubscribing (which does close a connection normally) will not cancel a _pending_ connection request (which the native timeout would do).

vsxed, 2020-04-28:
@werediver 
Thanks for your input!
Would it be possible if you could try to break this down for me?

As of now, my timer simply runs in parallel with the connection stream, and simply cancels the stream when the time runs out. Here's an extract from my bloc:

```
Timer _connectionTimeOutTimer;

...

void cancelConnTimeoutTimer() => _connectionTimeOutTimer?.cancel();

Future<void> cancelCurrentDeviceConnection() async {
    return await _connectToTargetDevice?.cancel()?.whenComplete(() {
      _currentDeviceId = null;
    });
  }

...
void handleBleConnection(BleConnectEvent event) async {
    await cancelDiscoveryScan();
    await cancelCurrentDeviceConnection();

    cancelConnTimeoutTimer();
    _connectionTimeOutTimer = Timer(Duration(seconds: ...), () async {
      await cancelCurrentDeviceConnection();

      /// Emit a failed connection state to update the ui
    });
    
    ...

    _connectToTargetDevice = _instance.connectToDevice(id: event.target.id).listen((update) {
      this.add(BleConnectionStateEvent(event.target, update));
    });
}
```

This code works (as of now) and uses the low-level callbacks from the library (`.cancel()` -> onCancel()), or am i wrong?

Also, pending connections seem to be correctly closed, since (also as of now) i didn't run into specific cases. Within the app, it is also possible to cancel a pending connection in the ui, which also works fine. 

Am i missing here something?

Btw, i am really thankful for all your help guys!

werediver, 2020-04-28:
Two highlights:

- you shouldn't need a timer to implement a "user-level" (so to say) connection timeout, just `stream.timeout()` would be enough
- Dart is not the low level I was alluding to, it's the _platform_ level; you can't trigger anything there from outside of the library

vsxed, 2020-04-29:
@werediver 
nice, didn't know that `.timeout(..)` was possible.
Need to dive in further into this, since a quick test showed, that the timeout will also trigger whenever the stream does not emit a state e.g. after a connection has been successfully established (since the lib does not periodically emit "connected" states. Otherwise a simple check within the timeout callback would also be enough. Will see later.

> Dart is not the low level I was alluding to, it's the platform level; you can't trigger anything there from outside of the library

I guess there's a misunderstanding here.
As far as i have seen [here](https://github.com/PhilipsHue/flutter_reactive_ble/blob/8bd525d1de7471b822d7660cf47aa2be373aaf01/lib/src/reactive_ble.dart#L375), cancelling the stream (`.connectToDevice(..).cancel()`) invokes the platform-level `disconnectFromDevice` method, which houses inside the corresponding android/ios PluginControllers.

werediver, 2020-04-29:
@vsxed You're right about `stream.timeout(…)`, I forgot it's not about the first event only ([ref](https://api.dart.dev/stable/2.8.0/dart-async/Stream/timeout.html)):

> Whenever more than `timeLimit` passes between two events …

Regarding the resources slipping: cannot confirm this claim now, because don't see potential leaks in the code (either eliminated or I remembered wrongly).

Still, there is a disadvantage of the out-of-the-library timeout on connection requests: it accounts for the time since the request has been _queued_, thus leaving less time for the actual connection, whereas the in-library timeout accounts for the time the BLE stack spends on establishing a connection.

Now, there is no significant difference between these two modes on iOS, because iOS is very capable of processing concurrent connection requests. On Android, though, the requests are _queued_ and processed one after another, which may lead to *huge* differences between the two modes (the [total] time since enqueuing till resolution vs. the [processing] time since an operation has been passed to the BLE stack till resolution).

This is important, if you're connecting to multiple devices.

remonh87, 2020-05-09:
For now I close this ticket if you still have questions about this feel free to  re-open it or log a new ticket

closed 2020-05-09 by remonh87