PhilipsHue/flutter_reactive_ble #62: How do I get a list of all services and characteristics for a device?

kaciula, 2020-05-08:
How do I get a list of all services and characteristics for a device? I've looked at the documentation but I could not find anything related to this. I see there is a way to read/write to a particular characteristic but no way to get all the services and characteristics.

Would appreciate any pointers.

werediver, 2020-05-08:
Listing the services and characteristics of a device is indeed not supported in the current version of the library. The most you can get at the moment is the `DiscoveredDevice.serviceData` dictionary, the keys of which correspond to the services for which the device advertises some data.

This functionality falls into the "BLE scanner" category, which is not the goal for the library, but will eventually be supported. When you're working with a specific device, your application should poses enough knowledge about its services and characteristics. If some of them are optional, their absence can be concluded from read/write errors.

Hope that resolves your question.

ch-muhammad-adil, 2020-07-14:
I also require these information because without these information we might not be able to read characteristic data , Here if you can expose BleGattProfile data after connection that is going to be helpful.

ch-muhammad-adil, 2020-07-14:
Basically I have a digital device which can detect  oxygen level of a person, so i want to get those information. What is the possible way? If listing of serviceIds and Characteristics ID is not supported then how can I read characteristics ?

werediver, 2020-07-14:
@ch-muhammad-adil You do not need to list services and characteristics of a known device (in most cases). Even if you have no documentation on a device, you can use a generic BLE scanner app (like [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) _once_ to get the IDs of services and characteristics of interest and use them in your app.

Regarding the support for listing services and characteristics available, the status is still as stated [above](https://github.com/PhilipsHue/flutter_reactive_ble/issues/62#issuecomment-625705884).

remonh87, 2020-09-18:
Issue is solved with #103 which will be available next release

closed 2020-09-18 by remonh87