PhilipsHue/flutter_reactive_ble #67: Long Device Name Getting Cutoff

ealanhill, 2020-05-29:
**Describe the bug**
I'm running into an issue with a long device name getting characters cut-off and I'm wondering if there's any debug steps I can take to work isolate where this is occurring. For example, I have a device who's name is: `11111111-2222-3333-4444-123456789012`, it looks like the last couple of characters are getting cut off when the scanning result info returns. This is what I will get back: `11111111-2222-3333-4444-12345`

**To Reproduce**
Steps to reproduce the behavior:
1. Start scan using `FlutterReactiveBle.scanForDevices(withServices: [])`
2. Listen for discovered devices and print to console device name:
```
listen((device) {
  developer.log('Name=${device.name}');
});
``
3. Observe a long name has the last couple characters are cut off.

**Expected behavior**
Full name of device is returned

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Samsung Galaxy Note8
 - OS: Android 8
 - Package version: v2.2.0

 - Device: Google Pixel 2
 - OS: Android 8
 - Package version: v2.2.0

**Peripheral device**
- Vendor, model: Raspberry Pi, Raspbian 2020-05-27
- Does it run a custom firmware / software: no

**Additional context**
This issue does not affect iOS

werediver, 2020-05-30:
@ealanhill I see you didn't tick the tried-doing-this-with-a-generic-BLE-scanner-app checkbox. Could you please confirm that a generic BLE scanner app (like nRF Connect mentioned in the template) reads out a long device name in full on the same device?

ealanhill, 2020-05-30:
@werediver Thank you for getting back to me! When I scan with nRF Connect, it gives me the version that is cutoff as well (using my Samsung Galaxy Note8). But, they label it as "Shortened Local Name".

Not having used nRF Connect is that a report?

It also seems like this is an Android specific limitation then?

werediver, 2020-05-30:
> Not having used nRF Connect is that a report?

It sure is, but in many cases—as this one—that provides useful insights.

> It also seems like this is an Android specific limitation then?

It looks like this, but it still may be worth checking whether there is an API or otherwise a way to get the full name.

@remonh87 Would you have a moment to check on this?

ealanhill, 2020-05-30:
@werediver @remonh87 Any more info I can provide to help look into this? Any steps I can take to check?

remonh87, 2020-05-31:
@ealanhill  this is an interesting question and it took me a bit of studying to find an answer to clarify the behaviour on Android. 

The max size of a field in an advertisement packet for BLE is 31bytes.  2 of those are used for the header so this leaves you with 29bytes for the device name.  The size of your device name  `11111111-2222-3333-4444-123456789012` is 36 bytes which is too long.  The size of the string `11111111-2222-3333-4444-12345` is exactly 29 bytes and this fits within the spec. For more details see for example: [source1](https://www.argenox.com/library/bluetooth-low-energy/ble-advertising-primer/) and [source2](https://devzone.nordicsemi.com/f/nordic-q-a/14/what-s-the-maximum-size-for-an-advertisement-package) .

I have really no clue why on iOS the name is not cut off that is a mystery to me. I did study the Android specs and also the implementation of the Android library we use and we do not cut off the name.

ealanhill, 2020-06-01:
@remonh87 Thank you so much for your research and answer! That helps quite a bit

remonh87, 2020-06-02:
You are welcome. For now I close this issue. Feel free to open it in case you found some other insights.

closed 2020-06-02 by remonh87