PhilipsHue/flutter_reactive_ble #68: Subscribing to characteristics crashes on iOS, works fine on Android

KarimAbdo, 2020-06-04:
First of all, thanks a lot for the library and the nice documentations :) 
Unfortunately, I couldn't get it working as expected on iOS.

**Describe the bug**
After successfully discovering, then connecting to a Bluetooth device (Blood pressure monitoring device), I could not subscribe and read the characteristics of the device.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectToDevice()`
2. subscribe to a protected characteristic `subscribeToCharacteristic()`
3. Observed a failure with the exception below

```
flutter: [BLE] subscribeToCharacteristic error is : PlatformException(flutter_reactive_ble.Central.(unknown context at $1071c0d30).Failure:2, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $1071c0d30).Failure error 2.), {})
```

**Expected behavior**
Reading blood pressure records similar to what happens on Android 
_(Receiving characteristics from the subscribeToCharacteristic function)_ 

**BlueTooth Service**
```
class BluetoothService implements DisposableService {
  static FlutterReactiveBle _ble = FlutterReactiveBle();

  MedicalDataService _medicalDataService = locator<MedicalDataService>();

  BehaviorSubject<ConnectionStateUpdate> connectionStateUpdate =
      BehaviorSubject();
  BehaviorSubject<List<DiscoveredDevice>> discoveredDeviceList =
      BehaviorSubject();
  BehaviorSubject<int> syncedDataCount = BehaviorSubject();

  StreamSubscription<ConnectionStateUpdate> _connectToDeviceSub;

  StreamSubscription<DiscoveredDevice> _scanForDevicesSub;

  StreamSubscription<List<int>> _subscribeToCharacteristicSub;

  Logger logger = Logger('BLE');

  BluetoothService() {
    print('bluetooth service init');
    _ble.statusStream.listen((event) {
      logger.debug('event = ${event}');
    });
  }

  @override
  void dispose() {
    _connectToDeviceSub?.cancel();
    _scanForDevicesSub?.cancel();
    _subscribeToCharacteristicSub?.cancel();
  }

  discoverDevices(BluetoothDevice device) {
    discoveredDeviceList.add([]);
    _scanForDevicesSub = _ble.scanForDevices(
        withServices: [Uuid.parse(device.uuid)]).listen((discoveredDevice) {
      if (discoveredDeviceList.hasValue &&
          !discoveredDeviceList.value.contains(discoveredDevice)) {
        discoveredDeviceList
            .add(discoveredDeviceList.value..add(discoveredDevice));
      }
    }, onError: (e) {
      logger.debug('discoverDevices error is : ${e}');
    });
  }

  connectToDevice(String deviceId, BluetoothDevice device) {
    List<Uuid> characteristicsUUIDs = [Uuid.parse(device.charId)];
    _connectToDeviceSub = _ble
        .connectToDevice(
            id: deviceId,
            servicesWithCharacteristicsToDiscover: {
              Uuid.parse(device.uuid): characteristicsUUIDs
            },
            connectionTimeout: Duration(seconds: 30))
        .listen((connectionState) {
      connectionStateUpdate.add(connectionState);
      logger.debug(
          'Status update at: ${DateTime.now().toIso8601String()} - state is: ${connectionState}');
    }, onError: (e) {
      logger.debug('connectToDevice error is : ${e}');
    });
  }

  readCharacteristic(DiscoveredDevice discoveredDevice, BluetoothDevice device,
      String charId) async {
    Uuid characteristicsUUID = Uuid.parse(charId);
    QualifiedCharacteristic characteristic = QualifiedCharacteristic(
        characteristicId: characteristicsUUID,
        serviceId: Uuid.parse(device.uuid),
        deviceId: discoveredDevice.id);

    final List<int> response =
        await _ble.readCharacteristic(characteristic).catchError((e) {
      logger.debug('readCharacteristic error is : ${e}');
    });
    return response;
  }

  subscribeToCharacteristic(DiscoveredDevice discoveredDevice,
      BluetoothDevice device, String charId) async {
    syncedDataCount.add(0);
    Uuid characteristicsUUID = Uuid.parse(charId);
    QualifiedCharacteristic characteristic = QualifiedCharacteristic(
        characteristicId: characteristicsUUID,
        serviceId: Uuid.parse(device.uuid),
        deviceId: discoveredDevice.id);

    _subscribeToCharacteristicSub = await _ble
        .subscribeToCharacteristic(characteristic)
        .listen((data) async {
      List<MedDataRecord> records = BluetoothDevice.parse(data, device);
      await records.forEach((record) => _medicalDataService.post((record)));
      syncedDataCount.add(++syncedDataCount.value);
    }, onError: (e, s) {
      syncedDataCount.add(-1);
      logger.debug('subscribeToCharacteristic error is : ${e}');
      logger.debug('subscribeToCharacteristic stack is : ${s}');
    });
    _subscribeToCharacteristicSub.resume();
  }

  Future writeCharacteristic(DiscoveredDevice discoveredDevice,
      BluetoothDevice device, String charId, List<int> byteArray) async {
    Uuid characteristicsUUID = Uuid.parse(charId);
    QualifiedCharacteristic characteristic = QualifiedCharacteristic(
        characteristicId: characteristicsUUID,
        serviceId: Uuid.parse(device.uuid),
        deviceId: discoveredDevice.id);

    var response = await _ble.writeCharacteristicWithResponse(characteristic,
        value: byteArray); // Could be wrong
    return response;
  }
}
```

**Smartphone / tablet**
 - Device: **Iphone 8 (Also tried with XR and iPad 11)**
 - OS: **iOS 13**
 - Package version: **[e.g. 1.0.0]**

**Peripheral device**
- Beurer, BM54, and other Bearer devices.
- Does it run a custom firmware: no

**Additional context**
- Android works fine with the same code
- Made sure the pairing is successful using vendor's app (Beurer Health Manager)

**Flutter Doctor**
Running `flutter doctor -v` result:

```
[✓] Flutter (Channel stable, v1.17.1, on Mac OS X 10.14.6 18G103, locale en-DE)
    • Flutter version 1.17.1 at /Users/buelenthacioglu/sdk/flutter
    • Framework revision f7a6a7906b (3 weeks ago), 2020-05-12 18:39:00 -0700
    • Engine revision 6bc433c6b6
    • Dart version 2.8.2
 
[!] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
    • Android SDK at /Users/buelenthacioglu/Library/Android/sdk
    • Platform android-29, build-tools 29.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)
    ! Some Android licenses not accepted.  To resolve this, run: flutter doctor --android-licenses
[✓] Xcode - develop for iOS and macOS (Xcode 11.3.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.3.1, Build version 11C504
    • CocoaPods version 1.8.4
[!] Android Studio (version 3.5)
    • Android Studio at /Applications/Android Studio.app/Contents
    ✗ Flutter plugin not installed; this adds Flutter specific functionality.
    ✗ Dart plugin not installed; this adds Dart specific functionality.
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)
[✓] IntelliJ IDEA Ultimate Edition (version 2019.3.2)
    • IntelliJ at /Applications/IntelliJ IDEA.app
    • Flutter plugin version 43.0.3
    • Dart plugin version 193.6015.53
[✓] Connected device (1 available)
    • Karim’s iPhone • 8418dbf9850aaef7c09c6c9760fbc1f02b5a74f8 • ios • iOS 13.3.1
```

werediver, 2020-06-04:
Hi Karim. I hope the library will serve you well in the end. It does have some limitation, though (e.g. it doesn't cover the generic-BLE-scanner-app use case).

I noticed something weird in your code here:

```dart
  connectToDevice(String deviceId, BluetoothDevice device) {
    List<Uuid> characteristicsUUIDs = [Uuid.parse(device.charId)];
    _connectToDeviceSub = _ble
        .connectToDevice(
            id: deviceId,
            servicesWithCharacteristicsToDiscover: {
              Uuid.parse(device.uuid): characteristicsUUIDs
            },
            connectionTimeout: Duration(seconds: 30))
...
```

The dictionary you're passing to `connectToDevice(⋯)` in `servicesWithCharacteristicsToDiscover:` parameter is supposed to map service IDs to their characteristic IDs. It looks like you're using the device ID instead of a service ID.

This works on Android, because `servicesWithCharacteristicsToDiscover` is not supported on Android (at least in the current implementation) and thus not taken into account.

On iOS this leads to not services and characteristic being discovered and therefore no operation (read/write/subscribe) can succeed for a connected device.

Please, check this.

KarimAbdo, 2020-06-04:
Thanks a lot Raman for the quick reply and the explanation.

I tried changing it to `Uuid.parse(device.uuid): [serviceUUID]` and also tried to remove it, but in the 3 cases, it produces the same exception.

This log could be useful:

```
flutter: [FlutterReactiveBLEApp] Starting to listen for devices...
flutter: qualified chars is 34EA0E38-170F-D810-3A1C-13B1EDE95E58  QualifiedCharacteristic(characteristicId: 00002a49-0000-1000-8000-00805f9b34fb, serviceId: 00001810-0000-1000-8000-00805f9b34fb, deviceId: 34EA0E38-170F-D810-3A1C-13B1EDE95E58)
flutter: [BLE] subscribeToCharacteristic error is : PlatformException(flutter_reactive_ble.Central.(unknown context at $105d04d30).Failure:2, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $105d04d30).Failure error 2.), {})
flutter: [BLE] subscribeToCharacteristic stack is : #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:156:18)
<asynchronous suspension>
#2      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:329:12)
#3      FlutterReactiveBle.subscribeToCharacteristic.<anonymous closure> (package:flutter_reactive_ble/src/reactive_ble.dart:448:14)
#4      Repeater.attach (package:flutter_reactive_ble/src/rx_ext/repeater.dart:36:48)
#5      _runGuarded (dart:async/stream_controller.dart:820:24)
#6      _BroadcastStreamController._subscribe (dart:async/broadcast_stream_controller.dart:215:7)
#7      _ControllerStream._createSubscription (dart:async/stream_controller.dart:833:19)
#8      _StreamImpl.listen (dart:async/stream_impl.dart:475:9)
#9      new _AddStreamState (dart:async/stream_controller.dart:900:34)
#10     new _StreamControllerAddStreamState (dart:async/stream_controller.dart:952:9)
#11     _StreamController.addStream (dart:async/stream_controller.dart:571:13)
```
and this 
```
flutter: Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at $105d04d30).Failure:2, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $105d04d30).Failure error 2.), {})
flutter: [FlutterReactiveBLEApp] Stopped listening for devices
flutter: [BLE] Status update at: 2020-06-04T13:14:21.314998 - state is: ConnectionStateUpdate(deviceId: 34EA0E38-170F-D810-3A1C-13B1EDE95E58, connectionState: DeviceConnectionState.connected, error: null)   - and device id is 34EA0E38-170F-D810-3A1C-13B1EDE95E58
flutter: [BLE] Status update at: 2020-06-04T13:14:51.505108 - state is: ConnectionStateUpdate(deviceId: 34EA0E38-170F-D810-3A1C-13B1EDE95E58, connectionState: DeviceConnectionState.disconnected, error: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}"))   - and device id is 34EA0E38-170F-D810-3A1C-13B1EDE95E58
```

werediver, 2020-06-04:
> I tried changing it to `Uuid.parse(device.uuid): [serviceUUID]` and also tried to remove it, but in the 3 cases, it produces the same exception.

I'm afraid you didn't get my message right.

You need to replace `Uuid.parse(device.uuid): [serviceUUID]` (or whatever it is now) with something like `service1UUID: service1characteristicsUUIDs` (and so on for other services, if/when you need any more).

KarimAbdo, 2020-06-04:
I tried that as well (basically that was the initial code (Just with bad naming)

werediver, 2020-06-04:
No, in the original code as provided in the very first message, instead of a service ID a device ID is used. Please, double-check. It should be a dictionary mapping service IDs to the corresponding characteristic IDs.

KarimAbdo, 2020-06-04:
I double-checked this,

I went back to this version and printed it, and logged it prints:
serviceId3:[charId2]  

serviceId3 is the service we want to access, and charId2 is the char we want to subscribe to.

`device.uuid` is just a misleading name, sorry about that.

BluetoothDevice device is our own model.

```
class UserBluetoothDevice {
  final String name;
  final String serviceId;  ---> previously called uuid (the misleading name)
  final String charId;
  final String writeCharId;
  final String picture;
  final String connectionHelperText;
  final String medicalDataType;
...
```

werediver, 2020-06-04:
Reading your logs I've got an impression that you're not stopping the scanning before connecting to devices. While it will work fine on iOS and may work on some Android devices, it's generally a really bad idea on Android, because of the Android BLE stack stability issues. Even on iOS the recommendation is to only run the discovery when it is necessary, because of energy saving reasons. That doesn't seem to relate to your main issue, though.

Are you sure you're subscribing to a characteristic only after the device state becomes `connected` (can't see it in the code)? That is important.

The latest released version 2.2.0 works fine in our apps, including subscriptions, so I'm pretty sure the library is not broken. I don't see what causes issues in your code, unfortunately.

@remonh87 Any ideas?

werediver, 2020-06-04:
Okay, the last suggestion. Triple-check whether you're discovering (`servicesWithCharacteristicsToDiscover:` in `connectToDevice(⋯)`) exactly the same service and characteristic you're later subscribing to.

KarimAbdo, 2020-06-05:
Thanks a lot Raman, will try that and also try to connect to another a device from another vendor to make sure where the problem is. will keep you updated

remonh87, 2020-06-24:
@KarimAbdo I will close the issue for now since I assume everything is working. Feel free to reopen it again if you still encounter issues

closed 2020-06-24 by remonh87

werediver, 2020-06-24:
@KarimAbdo Just in case you're still fighting your issue, check #76. It looks distantly resembling.

KarimAbdo, 2020-06-27:
Hi @werediver 

Thanks alot for getting back to me on this.
Unfortunately I couldn't really get this issue fixed, but my guess is that there are some problem connecting to certain devices with this package.

We switched to a different package (ble lib) and it worked with almost the same code.

Thanks alot for ur help anyways :)