PhilipsHue/flutter_reactive_ble #70: Improve our current external api

remonh87, 2020-06-10:
**Is your feature request related to a problem? (please describe)**
When we developed the library we decided to have a facade `FlutterReactiveBle` where a user can interact with. Intuitively it feels more natural for operations like write, read, subscribe etc to be operated on a device. This is also closer to the approach of iOS (CBPeripheral) and Android (BluetoothGatt).

**Describe the solution you'd like**
When we scan for devices we will get a `BlePeripheral` or `BluetoothDevice`. On this bluetooth device we are able to execute the following operations:

- connect and disconnect
- observeConnectionState
- read, write and subscribe to a characteristic
- request MTU size
- request connection priority  

Example pseudo code

```dart
final device =  await _ble.scanForDevices(withServices: serviceIds).first;

await device.readCharacteristic(serviceUuid: service, characteristicUuid: charUuid);

await device.requestMtu(mtuSize:250);
```

**Describe alternatives you've considered**
Let's use this ticket as a discussion how we can improve our API and to reflect on our current API.

remonh87, 2020-08-06:
@werediver  it was an interesting thought but I am not super convinced that my proposal is the way to go. I see some issues with the specific repeaters we use for multiple device connections (e.g connectionstateupdate and characteristic value) . Thereby I think the last refactoring we did for testability already provided good separation of concerns.  What are your thoughts?

werediver, 2020-08-06:
@remonh87 I'd say let's keep the current architecture for now, if your proposition doesn't "click into place" / doesn't look transparent enough to implement. If with time we see better how this or another approach can improve the plug-in architecture, we can try it at any moment with no problems.

closed 2020-08-11 by remonh87