PhilipsHue/flutter_reactive_ble #73: No catchable exception when BT is disabled in scanRepeater

mikes222, 2020-06-16:
**Describe the bug**
When BT is disabled in Android and the scan is started there is an exception internally but the exception is not handled to the calling program and therefore cannot be handled by the app.

**To Reproduce**
Steps to reproduce the behavior:
1. In Android switch off BT
2. start _scanner.startScan(null); from the example program
3. Check the console for failures

**Expected behavior**
We need a stream event for exceptions and some additional information about the type of exception

**Additional context**

```
I/flutter (21735): ErrorValue: Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Bluetooth disabled (code 1)")
I/flutter (21735): #0      Result.dematerialize.<anonymous closure> (package:flutter_reactive_ble/src/model/result.dart:22:13)
I/flutter (21735): #1      Result.iif (package:flutter_reactive_ble/src/model/result.dart:34:21)
I/flutter (21735): #2      Result.dematerialize (package:flutter_reactive_ble/src/model/result.dart:15:28)
I/flutter (21735): #3      FlutterReactiveBle.scanForDevices.<anonymous closure>.<anonymous closure> (package:flutter_reactive_ble/src/reactive_ble.dart:274:49)
I/flutter (21735): #4      _MapStream._handleData (dart:async/stream_pipe.dart:229:31)
I/flutter (21735): #5      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:166:13)
I/flutter (21735): #6      _rootRunUnary (dart:async/zone.dart:1192:38)
I/flutter (21735): #7      _CustomZone.runUnary (dart:async/zone.dart:1085:19)
I/flutter (21735): #8      _CustomZone.runUnaryGuarded (dart:async/zone.dart:987:7)
I/flutter (21735): #9      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter (21735): #10     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:266:7)
I/flutter (21735): #11     _ForwardingStreamSubscription._add (dart:async/stream_pi

```

remonh87, 2020-06-16:
Thanks for reporting and a good catch.  You are right we should propagate the error in the current stream. 

I can reproduce the behavior and will see if I can find a fix for it.

remonh87, 2020-06-18:
Issue will be fixed in v2.3.0 expect to release it coming days

closed 2020-06-18 by remonh87

remonh87, 2020-06-22:
@mikes222 we released v.2.3.0 that includes the fix for your issue

PR #75 (2020-06-18): fix propagating stream error in devicescanner
Fixes #73