PhilipsHue/flutter_reactive_ble #76: Small inconsistency between iOS and Android with short/long UUID

xvrh, 2020-06-19:
Full disclosure: I don't really understand what I'm talking about :-) I'm just reporting this in case it helps someone.

**Describe the bug**
I noticed a small difference between Android and iOS when registering a characteristic for Notification (`subscribeToCharacteristic`).

My device "somehow" uses "short UUID" (16 bits) but I was using the long form (128bits) in my code (`QualifiedCharacteristic`).
It works for Read and Write on both Android and iOS. But for "Subscribe" it only works for Android. 

On iOS I had no error but I never received the data. 

With the short Uuid, the notification worked on both platform.

**Expected behavior**
It should work the same on both Android and iOS. The iOS code should probably map the short Uuid (sent by the device?) to the long form registered on the phone?

**Smartphone / tablet**
 - Device: iPhone 11 and Moto G5
 - OS: iOS 13, Android 8.1
 - Package version: 2.2.0

**Peripheral device**
- Vendor, model: ESP32
- Does it run a custom firmware / software: custom firmware based on ESP-IDF v4

werediver, 2020-06-22:
@xvrh I imagine it wasn't easy to figure this out, thank you for sharing your findings.

The library doesn't do any transformations on service or characteristic IDs and at the moment I am not sure there is a strict rule of mapping short IDs to long or vice versa that would consistently work in all cases with all devices.

I'm afraid, your solution is a solution for a particular kind of devices and may not work with others. If you have reasons to think otherwise, I'd appreciate if you provide your references.

xvrh, 2020-06-23:
Hi Raman, Thanks for the answer.

What surprise me, is that it worked with the plugin "flutter_blue". 
I will try to dig into this a bit to understand where the data is lost.

remonh87, 2020-07-24:
@xvrh any luck on understanding why the data is lost?

xvrh, 2020-07-24:
@remonh87 sorry, not yet.
I will close this issue until I have more concrete information to report :-)

Thanks.

closed 2020-07-24 by xvrh

LiangHuangBC, 2021-12-02:
Any update on this issue?  I have same issue, subscribeToCharacteristic is only working on android, there is no data on IOS. Not sure what you mean by shot UUID or long. Anyway to pass this around?

> @remonh87 sorry, not yet. I will close this issue until I have more concrete information to report :-)
> 
> Thanks.