PhilipsHue/flutter_reactive_ble #79: Crash when disconect

RiuHDuo, 2020-06-24:
**Describe the bug**
The lib crashed when i shutdown the connected peripheral.

**Expected behavior**
Crashed Stack
```
  Process: com.wellpay.deoxys, PID: 29020
    io.reactivex.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
        at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.dispose(ObservableFlatMap.java:313)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.dispose(ObservableReplay.java:271)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableRefCount.timeout(ObservableRefCount.java:137)
        at io.reactivex.internal.operators.observable.ObservableRefCount.cancel(ObservableRefCount.java:104)
        at io.reactivex.internal.operators.observable.ObservableRefCount$RefCountObserver.dispose(ObservableRefCount.java:233)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.mixed.CompletableAndThenObservable$AndThenObservableObserver.dispose(CompletableAndThenObservable.java:86)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.mixed.SingleFlatMapObservable$FlatMapObserver.dispose(SingleFlatMapObservable.java:84)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.mixed.SingleFlatMapObservable$FlatMapObserver.dispose(SingleFlatMapObservable.java:84)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.dispose(ObservableFlatMap.java:588)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.disposeAll(ObservableFlatMap.java:510)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.dispose(ObservableFlatMap.java:310)
        at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.dispose(ObservableDoOnEach.java:79)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.disposeAll(ObservableFlatMap.java:504)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.dispose(ObservableFlatMap.java:310)
        at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.dispose(ObservableObserveOn.java:146)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.observers.LambdaObserver.dispose(LambdaObserver.java:102)
        at com.signify.hue.flutterreactiveble.channelhandlers.CharNotificationHandler.unsubscribeFromNotifications(CharNotificationHandler.kt:46)
        at com.signify.hue.flutterreactiveble.PluginController.stopNotifications(PluginController.kt:231)
        at com.signify.hue.flutterreactiveble.PluginController.access$stopNotifications(PluginController.kt:25)
        at com.signify.hue.flutterreactiveble.PluginController$pluginMethods$11.invoke(PluginController.kt:37)
        at com.signify.hue.flutterreactiveble.PluginController$pluginMethods$11.invoke(PluginController.kt:25)
        at com.signify.hue.flutterreactiveble.PluginController.execute$flutter_reactive_ble_debug(PluginController.kt:75)
        at com.signify.hue.flutterreactiveble.ReactiveBlePlugin.onMethodCall(ReactiveBlePlugin.kt:43)
        at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:226)
```

**Smartphone / tablet**
 - Device: Pixel4
 - OS: Android 11
 - Package version: 2.2

remonh87, 2020-06-24:
This is a common issue in RXJava2, see https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling for further details. Also on our readme I have described a workaround how to handle it. The fundamental problem lays in the Android threading model. 

I will investigate the issue later to see if I can look for a more gracious solution but I haven't high hopes.

remonh87, 2020-07-24:
@RiuHDuo were you able solving the issue?

closed 2020-07-27 by remonh87

chris-otani, 2021-08-19:
@remonh87 , I'm experiencing the same error 

> Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)")

It sounds like the clear gatt cache method in the readme might fix this. Can you advise where this should be placed in the code? Does it go write before disconnecting from the module?

await flutterReactiveBle.clearGattCache(foundDeviceId);