PhilipsHue/flutter_reactive_ble #80: iOS render overflow in example Scan for devices Enter UUID text

andrewknockin, 2020-06-28:
Firstly very many thanks for a wonderfully presented, designed and documented library.

**Describe the bug**
On iPhones with screen size 750 x 1334 pixels the example Scan for Devices screen has a line of text:
Enter a UUID above and tap start to begin scanning
which overflows to the right losing the word 'scanning'.
This causes an exception by the rendering library with a yellow warning on screen and the exception reported to console, but does not stop the app running and continuing.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the example on an iPhone6, iPhone6s, iPhone7, iPhone8, iPhoneSE with horizontal screen size 750 pixels.

**Expected behavior**
No text overflow.

**Additional context**
iOS13, Xcode 11.5

The console report is:

════════ Exception caught by rendering library 
The following assertion was thrown during layout:
A RenderFlex overflowed by 76 pixels on the right.

The relevant error-causing widget was
    Row 
lib/…/ui/device_list.dart:117
The overflowing RenderFlex has an orientation of Axis.horizontal.
The edge of the RenderFlex that is overflowing has been marked in the rendering with a yellow and black striped pattern. This is usually caused by the contents being too big for the RenderFlex.

Consider applying a flex factor (e.g. using an Expanded widget) to force the children of the RenderFlex to fit within the available space instead of being sized to their natural size.
This is considered an error condition because it indicates that there is content that cannot be seen. If the content is legitimately bigger than the available space, consider clipping it with a ClipRect widget before putting it in the flex, or using a scrollable container rather than a Flex, like a ListView.

The specific RenderFlex in question is: RenderFlex#3e75a relayoutBoundary=up4 OVERFLOWING

andrewknockin, 2020-06-29:
Suggest change the text (device_list.dart line 121) from
Enter a UUID above and tap start to begin scanning
 to
Enter a UUID above then tap Scan
since the button is labelled Scan not start.

andrewknockin, 2020-06-29:
Also, this section of code in device_list.dart requires the 'control-flow-collections' experiment to be enabled to allow the 'if' beginning line 123 to be valid.

remonh87, 2020-06-29:
@andrewknockin Thanks for the good explanation and reporting the issues  I will have a look at it for the coming release. Feel free to create a PR yourself if you are up to it :)

remonh87, 2020-07-01:
merged fix into master. Will ship it along side with version 2.4.0 that will be released in the coming weeks

closed 2020-07-01 by remonh87