PhilipsHue/flutter_reactive_ble #86: Error 'Kotlin reflection is not available' when sending writeCharacteristicWithResponse

mbn18, 2020-07-09:
**Describe the bug**
Everytime a `writeCharacteristicWithResponse` is dispatched I see in the log:
```
D/PluginController$executeWriteAndPropagateResultToChannel( 8898): Value succesfully written, function writeCharacteristicWithResponse (Kotlin reflection is not available)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Dispatch `writeCharacteristicWithResponse`

**Smartphone**
 - Device: Samsung A107F
 - OS: Android 10
 - Package version: 2.3.0

**Peripheral device**
- Vendor, model: raspberry pi 3 b+
- Go code implementing [go-ble](https://github.com/go-ble/ble)

**Additional context**
It seems that everything works correctly even though the error float

remonh87, 2020-07-09:
Thanks for bringing this to our attention. I suspect this has to do with the logging of the operation.  
I actually want to get rid of this verbose logging on Android and create a better logger.

andrewknockin, 2020-07-19:
I am using WithoutResponse and see something very similar.
The message has actually proved to me that I am writing commands, because since the upgrade to Flutter 1.17.5 today, the answers back from my device no longer appear at the characteristic listen method.
So please don't remove logging completely.

remonh87, 2020-07-28:
Improved the logging and made it consistent for iOS and Android. This will be available in version 2.4.0 that I hope to publish today

closed 2020-07-28 by remonh87