PhilipsHue/flutter_reactive_ble #87: Scanning parameters are not set

Errichamonda, 2020-07-13:
**Describe the bug**
When I try to scan for devices it always throws an Exception. The demo application works fine but when I try to use it on my app it doesn't work.

**To Reproduce**
Initialize and then scan with `scanForDevices`

**Expected behavior**
Print Log a list of MAC addresses of the devices.

- [x] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: MiPAD4Plus
 - OS: Android 10
 - Package version: 2.3.0 | git master

remonh87, 2020-07-13:
@Errichamonda  thanks for reporting the issue. Can you supply me the code that you are using for calling the `scanForDevices`? Make sure that the parameter `withServices` has a value as documented in the code.

Errichamonda, 2020-07-13:
It's a simple Stateful Widget
```
class Screen extends StatefulWidget {
  Screen();

  @override
  _ScreenState createState() => _ScreenState();
}

class _ScreenState extends State<Screen> {
  FlutterReactiveBle _ble = FlutterReactiveBle();
  @override
  Widget build(BuildContext context) {
    return Scaffold(
      body: Center(
        child: Column(
          mainAxisSize: MainAxisSize.max,
          mainAxisAlignment: MainAxisAlignment.center,
          children: [
            RaisedButton(
              onPressed: () {
                widget._ble.scanForDevices(withServices: [], scanMode: ScanMode.lowPower, requireLocationServicesEnabled: true).listen((event) {
                  print(event.id);
                }).onError((handleError) {
                  print(handleError);
                });
              },
              child: Text("Scan"),
            ),
          ],
        ),
      ),
    );
  }
}
```

remonh87, 2020-07-13:
ah ok I see what is the issue now. You execute a stream subscription at on build which can potentially be called 60 times a second. Because of this you will run in a race condition on the Android layer. I would suggest to look to our example app. We execute the startscan on onTap but maintaining the subscription is done outside the build method. A simple solution can already be:
```dart
class Screen extends StatefulWidget {
  Screen();

  @override
  _ScreenState createState() => _ScreenState();
}

class _ScreenState extends State<Screen> {
  FlutterReactiveBle _ble = FlutterReactiveBle();

void startScan() {
   _ble.scanForDevices(withServices: [], scanMode: ScanMode.lowPower, requireLocationServicesEnabled: true).listen((event) {
     print(event.id);
     }).onError((handleError) {
       print(handleError);
    });
}
  @override
  Widget build(BuildContext context) {
    return Scaffold(
      body: Center(
        child: Column(
          mainAxisSize: MainAxisSize.max,
          mainAxisAlignment: MainAxisAlignment.center,
          children: [
            RaisedButton(
              onPressed: startScan,
              child: Text("Scan"),
            ),
          ],
        ),
      ),
    );
  }
}
```

Errichamonda, 2020-07-13:
Hi, I've tried the code that you sent me and I'm still getting the same error.
I've noticed that if I move from compileSDK 30 to 28 it works.

remonh87, 2020-07-14:
That is very interesting... I updated the example app to compileSDK 30 and I have no issues.  I use the exact same code as what I explained above and it works without raising an issue. I tested it both on Samsung and a google pixel as wel a galaxy tab.  What I am thinking is it could be a device specific issue. Are you in the position of testing it on another android device? Also which version of Flutter are you using? I am using , v1.17.3

Errichamonda, 2020-07-14:
```
Flutter (Channel beta, 1.19.0-4.3.pre, on Microsoft Windows [Versione 10.0.19041.329], locale it-IT)
    • Flutter version 1.19.0-4.3.pre at C:\flutter
    • Framework revision 8fe7655ed2 (12 days ago), 2020-07-01 14:31:18 -0700
    • Engine revision 9a28c3bcf4
    • Dart version 2.9.0 (build 2.9.0-14.1.beta)
```
I've tested on a OnePlus 6 with compileSDK 30 and it seems to work. Maybe it's something about Xiaomi software that causes the error. If I still encounter this error I'll reopen this issue. Thanks for the help.

omarkamal, 2020-07-30:
Hi, I'm having pretty much the same issue here. I was using a provider/get_it/mvvm architecture but I simplified it to the same code snipped shared above. I'm on Flutter 1.17.5 (the current stable). 

I've tried this on 3 different Android devices, but no luck. 

Here are my observations : 
1. If I leave the permission off (using the exact same code as above), the first button press says no location, the second button press says scanning parameters are not set
2. If explicitly go to the app info and give it location permission, it works like a charm
3. Even the example app compiled on my device throws the error in the image attached : 

[IMAGE 2020-07-30 21:18:49](https://user-images.githubusercontent.com/1854181/88944367-44470600-d2aa-11ea-9e11-31c1095dbd0a.jpg)

Help?

remonh87, 2020-07-30:
@omarkamal we require location permissions since this is the safest way of scanning.  You can avoid this requirement by:

` _ble.scanForDevices(withServices: serviceIds, requireLocationServicesEnabled: false)` but this is at your own risk since Nexus devices need location services for scanning. I will reopen the issue because I am curious what happens on the second tap

omarkamal, 2020-07-30:
Hey, thanks for your response!

But that's the odd thing I noticed - even when I didn't explicitly set that flag, it never asked me for the location permission. 

It was denied by default and I had to open app info and allow permissions there. Even setting the Bluetooth permissions on the project's own androidManifest didn't throw up a location prompt. 

Tried this on a pixel 2 xl and an mi a2 both on Android 10. 

I'll dig into the double button press a bit more to make sure it's nothing silly I'm doing on my end!

remonh87, 2020-07-30:
That is intended. Before scanning you need to make sure that the status of your ble is set to `Ready`.  See `ble_status_monitor` in the example app for a simple implementation. We do not handle automatic asking for permissions since it is up to the user to decide how they want to handle it. For example https://pub.dev/packages/permission_handler provides a good way of handling this.

closed 2020-07-31 by remonh87