PhilipsHue/flutter_reactive_ble #88: Ble Undeliverable Exception Documentation

0p3r4t0r, 2020-07-15:
**Is your feature request related to a problem? (please describe)**
The ReadMe section for [Ble Undeliverable Exception](https://github.com/PhilipsHue/flutter_reactive_ble#ble-undeliverable-exception) should contain a **complete** working code sample and some more detailed instructions.

**Describe the solution you'd like**
To catch the `Ble Undeliverable` exception in Java you can do the following . . .
1. Add `implementation "com.polidea.rxandroidble2:rxandroidble:1.11.1"` to the dependencies section of your **build.gradle** file ([source](https://github.com/Polidea/RxAndroidBle#gradle)).

2. Then add the following to **MainActivity.java**
```java
import io.flutter.embedding.android.FlutterActivity;
import androidx.annotation.NonNull;
import io.flutter.embedding.android.FlutterActivity;
import io.flutter.embedding.engine.FlutterEngine;
import com.polidea.rxandroidble2.exceptions.BleException;
import io.reactivex.exceptions.UndeliverableException;
import io.reactivex.plugins.RxJavaPlugins;

public class MainActivity extends FlutterActivity {
    public void configureFlutterEngine(@NonNull FlutterEngine flutterEngine) {
        super.configureFlutterEngine(flutterEngine);
        RxJavaPlugins.setErrorHandler(throwable -> {
            if (throwable instanceof UndeliverableException && throwable.getCause() instanceof BleException) {
                System.out.println('!' * 80);
                System.out.println("Caught UndeliverableException from flutter_reactive_ble.");
                System.out.println('!' * 80);
                return; // ignore BleExceptions as they were surely delivered at least once
            }
            throw new RuntimeException("Unexpected Throwable in RxJavaPlugins error handler", throwable);
        });
    }
}
```

**Additional context**
As for me, my experience with android is limited and so it took me a long time to build a complete working version from the code you had in your README. Also, you can find some helpful examples here in the [RxAndroidBle library examples folder](https://github.com/Polidea/RxAndroidBle/tree/13cb034751f08b677abe1bc8fcda6b28c34e924b/sample/src/main/java/com/polidea/rxandroidble2/sample).

remonh87, 2020-07-15:
@0p3r4t0r  thanks for reporting. I indeed see that I forgot to add the need for adding the polidea library in this example.  Regarding the rest of the code you are working in java and my example was written in Kotlin. Will make that more clear as well. For the rest the issue is not on the RxAndroid library but more on the RXJava side so adding the RXAndroidBLE examples do not add that much value.

0p3r4t0r, 2020-07-15:
How about providing some links to the relevant sections of the RxAndroid library? That way you wouldn't clutter your docs and you could probably save your users some Googling.

Something like . . .
* Also see [RxAndroidBle](https://github.com/Polidea/RxAndroidBle) for details and check [RxAndroidBle: Examples](https://github.com/Polidea/RxAndroidBle/tree/13cb034751f08b677abe1bc8fcda6b28c34e924b/sample/src/main/java/com/polidea/rxandroidble2/sample) for an example implementation.

remonh87, 2020-07-24:
improved the readme and it is now available on master.

closed 2020-07-24 by remonh87

amalpersonal, 2023-06-14:
@0p3r4t0r Appreciate all the info you put in. Is it possible for you to share the fully working version of the sample app that takes care of the Undeliverable exception? If someone already has that solved, can you please point me to that version? 

I am not too familiar with the Android/Java/Kotlin infrastructure to debug the problem I am having. I tried making changes to build.gradle and MainActivity.java but my code does not build. I get different sorts of errors depending on things that I try. E.g. with your suggestion as is I get: "Could not find method implementation() for arguments [com.polidea.rxandroidble2:rxandroidble:1.11.1]" 

Can you please let me know how to set up the Flutter environment so that the Android/Java with gradle is able to find all the necessary symbols and/or packages?