PhilipsHue/flutter_reactive_ble #89: Add permission documentation for iOS & Android

vanlooverenkoen, 2020-07-22:
Because we need to add permissions to android & iOS before we can get this to work. 

This should be added to the readme as well.

remonh87, 2020-07-22:
Hi @vanlooverenkoen  makes sense will clarify the readme on this topic.

vanlooverenkoen, 2020-07-22:
I was trying to get the https://github.com/BeMacized/xiaomi_scale package to work. 
I got your example working in my project but couldn't get the xiaomi_scale package to work. 
The example from xiaomi_scale was requesting location permission as well. But because I did not provide the correct info.plist data the app crashed.

I just got that package working. But I think flutter_reactive_ble should have a section in the readme that clarifies what is need and what is already done by the package itself.

remonh87, 2020-07-24:
improved the readme and it is now available on master

closed 2020-07-24 by remonh87