PhilipsHue/flutter_reactive_ble #91: Throw exception on subscription to non-readable characteristic

tgmoo, 2020-07-23:
We spent a long time troubleshooting why we stopped receiving data from the BLE device.  Turned out to be that a firmware update erroneously changed the characteristic  to "write" only.  Perhaps trying to subscribe to a characteristic that didn't allow reading should throw an exception because it cannot possibly ever actually receive any data.

werediver, 2020-07-24:
@tgmoo Please, use the suggested template when opening an issue. Even though some questions in it may be (or seem) irrelevant, some other would save time to everyone.

On iOS the corresponding checks are implemented:

```swift
func turnNotifications(_ state: OnOff, for qualifiedCharacteristic: QualifiedCharacteristic, completion: @escaping CharacteristicNotifyCompletionHandler) throws {
    let characteristic = try resolve(characteristic: qualifiedCharacteristic)

    guard [CBCharacteristicProperties.notify, .notifyEncryptionRequired, .indicate, .indicateEncryptionRequired]
            .contains(where: characteristic.properties.contains)
    else { throw Failure.notificationsNotSupported(qualifiedCharacteristic) }

    ...
```

Which makes me think you are only working with Android.

@remonh87 Could you please clarify the situation on Android side?

werediver, 2020-07-24:
As a separate point: the "read" permission has nothing to do with notifications and indications (subscribing to a characteristic). The relevant permissions are shown in the code snippet in the answer above.

remonh87, 2020-07-24:
Clarification on the Android side:  we check if the characteristic supports notification by checking this propery: [official android doc](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic#PROPERTY_NOTIFY).  If not we setup an indication and there it can possibly go wrong because if the characteristic do not support inidications we should provide an error. I will improve this part. However not sure if this will fix your problem

tgmoo, 2020-07-24:
We definitely had a problem in our device.  The characteristic might have
had notify set without read, which makes no sense.  Without read being set
there is no way that the subscription would ever receive any data, which I
was suggesting might warrant an exception or some sort of warning.  I have
figured this out, so will not ever waste much time on this specific problem
again, but others may.

Yes, we have only been able to test that far on Android.  The iOS app
crashes before it gets that far, saying something about _ble not being
found.  Because of that the team is concentrating on getting basic
functionality up on Android first.  I'd like to submit that separately as
it's own issue.

werediver, 2020-07-24:
> The characteristic might have had notify set without read, which makes no sense.  Without read being set
there is no way that the subscription would ever receive any data, ...

@tgmoo This is not correct. A notification or indication carries data on its own. It makes total sense for a characteristic to have "notify" or "indicate" flags without "read". It's basically reactive approach vs. polling.

werediver, 2020-10-07:
The root of the issue appears to be in the understanding of the BLE notification/indication mechanism by the original poster. I hope, we provided enough information to help understanding the subject mechanism.

closed 2020-10-07 by werediver