PhilipsHue/flutter_reactive_ble #94: How to connect multiple different devices at the same time?

yihuishou, 2020-08-04:
**Is your feature request related to a problem? (please describe)**
Connect multiple devices and write different information to different devices at the same time

**Describe the solution you'd like**
There is no information about multiple device connection writes in the documentation and examples

**Describe alternatives you've considered**
Nothing

**Additional context**
Nothing

remonh87, 2020-08-06:
@yihuishou as long as you make sure you do not close the connection by terminating the stream you can connect to and execute operations on multiple devices the same way as we documented for one device. The library will make sure it handles the BLE operations like desired by the specific operating system. If you are missing something can you please elaborate more on what you would like to see in the docs?

yihuishou, 2020-08-17:
After commenting out the content of closing the stream in the example, it is really possible to connect multiple devices with Bluetooth. Thank you 😃

closed 2020-08-18 by remonh87

Ilya-Vys, 2023-03-21:
> After commenting out the content of closing the stream in the example, it is really possible to connect multiple devices with Bluetooth. Thank you 😃

I facing a first peripherial disconnecting when i start to searching second one with scan. Which line i have to comment?