PhilipsHue/flutter_reactive_ble #97: When I Turn off BLE or disconnect a device my app crashes on Flutter 1.20.1 (but not on 1.17.5)

lwinged, 2020-08-10:
When I Turn off BLE or disconnect a device my app crashes on Flutter 1.20.1 (but not on 1.17.5)

Hello, 

Firstly, thank you for your great job :) 

I’ve an issue on flutter 1.20.1 but not on 1.17.5, I don't know if the issue comes from the flutter update or the library ?

My issue : When I turn off the BLE or if I disconnect manually/physically (not via code) a BLE device, my app crashes on flutter 1.20.1 (but not on 1.17.5).

I don’t know how to handle this exception, do you have any idea ?

Thank you very much

info :
Device :  Samsung Galaxy A10
OS : Android 10

**com.polidea.rxandroidble2.exceptions.BleDisconnectedException** **com.polidea.rxandroidble2.exceptions.BleAdapterDisabledException**

`
io.reactivex.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status -1 (UNKNOWN)
E/AndroidRuntime(21736): 	at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:704)
E/AndroidRuntime(21736): 	at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:701)
E/AndroidRuntime(21736): 	at io.reactivex.internal.observers.LambdaObserver.onError(LambdaObserver.java:77)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.checkTerminated(ObservableObserveOn.java:281)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:172)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:255)
E/AndroidRuntime(21736): 	at io.reactivex.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:124)
E/AndroidRuntime(21736): 	at android.os.Handler.handleCallback(Handler.java:883)
E/AndroidRuntime(21736): 	at android.os.Handler.dispatchMessage(Handler.java:100)
E/AndroidRuntime(21736): 	at android.os.Looper.loop(Looper.java:237)
E/AndroidRuntime(21736): 	at android.app.ActivityThread.main(ActivityThread.java:7948)
E/AndroidRuntime(21736): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(21736): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
E/AndroidRuntime(21736): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1075)
E/AndroidRuntime(21736): Caused by: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status -1 (UNKNOWN)
E/AndroidRuntime(21736): 	at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter$3.apply(DisconnectionRouter.java:53)
E/AndroidRuntime(21736): 	at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter$3.apply(DisconnectionRouter.java:50)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:57)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableFilter$FilterObserver.onNext(ObservableFilter.java:52)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver$DelayErrorInnerObserver.onNext(ObservableConcatMap.java:506)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:62)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableRefCount$RefCountObserver.onNext(ObservableRefCount.java:227)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservablePublishAlt$PublishConnection.onNext(ObservablePublishAlt.java:177)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onNext(ObservableUnsubscribeOn.java:60)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onNext(ObservableSubscribeOn.java:58)
E/AndroidRuntime(21736): 	at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onNext(ObservableCreate.java:66)
E/AndroidRuntime(21736): 	at com.polidea.rxandroidble2.RxBleAdapterStateObservable$1$1.onReceive(RxBleAdapterStateObservable.java:67)
E/AndroidRuntime(21736): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1646)
E/AndroidRuntime(21736): 	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
E/AndroidRuntime(21736): 	... 7 more
E/AndroidRuntime(21736): Caused by: com.polidea.rxandroidble2.exceptions.BleAdapterDisabledException
E/AndroidRuntime(21736): 	... 21 more
`

remonh87, 2020-08-10:
@lwinged  thanks for reporting the issue. Can you try version 2.3.0 and see if the issue still occurs there?

remonh87, 2020-08-11:
@lwinged shipped `2.4.0+1` which contains hotfix for the issue

PR #98 (2020-08-11): fix for crash on notifications when device gets disconnected
#92 revealed some in proper error handling on the notifications.

Fixes #97
closed 2020-08-11 by remonh87