PhilipsHue/flutter_reactive_ble #100: iOS always sends the same command, Android works OK

andrewknockin, 2020-08-11:
Following my problems reported in #95 my code works for Android, but in iOS it sends the first command correctly, then every following command is just the first command repeated. At the end of my connect routine, I need to get different state information from the bluetooth device so I send a set of commands:
```
void sendInitialCommands() async {
    print("====sendInitialCommands");
    sendString('y');
    sendString('x');
    sendString('h');
    print("=======================");
  }

  sendString(String dataStr) async {
    List<int> dataOut = asciiCodec.encode(dataStr);
    print("Sending " + asciiCodec.decode(dataOut));
    await _ble.writeCharacteristicWithResponse(txCharacteristic, value: dataOut);
  }
```
on Android I correctly get answers to those 3 commands:
```
D/BluetoothGatt(15802): connect() - device: A4:CF:12:78:10:3E, auto: true
D/BluetoothGatt(15802): registerApp()
D/BluetoothGatt(15802): registerApp() - UUID=18f9a898-54d2-4fcf-9960-29562c60ef2e
D/BluetoothGatt(15802): onClientRegistered() - status=0 clientIf=8
I/flutter (15802): ====SendInitialCommands
I/flutter (15802): Sending y
I/flutter (15802): Sending x
I/flutter (15802): Sending h
/flutter (15802): =======================
D/BluetoothGatt(15802): onClientConnectionState() - status=0 clientIf=8 device=A4:CF:12:78:10:3E
D/BluetoothGatt(15802): discoverServices() - device: A4:CF:12:78:10:3E
I/flutter (15802): connectToAdvertisingDevice error: TimeoutException after 0:00:06.000000: Future not completed
D/BluetoothGatt(15802): onConnectionUpdated() - Device=A4:CF:12:78:10:3E interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(15802): onSearchComplete() = Device=A4:CF:12:78:10:3E Status=0
D/BluetoothGatt(15802): onConnectionUpdated() - Device=A4:CF:12:78:10:3E interval=30 latency=0 timeout=500 status=0
D/BluetoothGatt(15802): setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
I/flutter (15802): Received: [y 10]
I/flutter (15802): Received: [x 8]
I/flutter (15802): Received: [h 160]

```
However, on iOS it reports the first command being correctly sent but subsequent commands only re-trigger the first command. I can see on the bluetooth device that the same command is transmitted repeatedly. Also, the Gatt reporting shown above for Android is missing on iOS.
```
flutter: Status update at: 2020-08-11T17:34:04.876624 to: ConnectionStateUpdate(deviceId: 0E76782B-8B93-89F0-B597-8F2963FAE171, connectionState: DeviceConnectionState.connected, error: null)
flutter: ====SendInitialCommands
flutter: Sending y
flutter: Sending x
flutter: Sending h
flutter: =======================
flutter: Received: [y 10]
flutter: Received: [y 10]
flutter: Received: [y 10]
```
How can I debug what is happening within
   await _ble.writeCharacteristicWithResponse(txCharacteristic, value: dataOut);
or what am I doing wrong?

werediver, 2020-08-12:
@andrewknockin I wouldn't call your code correct (I wouldn't expect the result you're describing anyway, but let's eliminate issues one by one).

Your `sendString(·)` function is asynchronous, but you're not waiting for its call to complete, before making another:

```dart
void sendInitialCommands() async {
  print("====sendInitialCommands");
  sendString('y');
  sendString('x');
  sendString('h');
  print("=======================");
}
```

should be 

```dart
Future<void> sendInitialCommands() async {
  print("====sendInitialCommands");
  await sendString('y');
  await sendString('x');
  await sendString('h');
  print("=======================");
}
```

I suggest you to check whether the corrected code works for you, before we look into the other possible causes of the issue.

andrewknockin, 2020-08-12:
Thank you very much @werediver. Adding the await solves the problem.

closed 2020-08-12 by andrewknockin