PhilipsHue/flutter_reactive_ble #101: example can't connect to iphone from android

Matrix-Zhang, 2020-08-13:
**Describe the bug**
the example app can scan found my iphone, but can't connect to iphone from my android

**Expected behavior**
should connected to my iphone

- [ yes ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: iphone xs max, redmi 10x
 - OS: ios 13, MIUI 12.0.4
 - Package version: [e.g. 1.0.0]

remonh87, 2020-08-13:
@Matrix-Zhang  I need more information on it can you provide me the device logs with the error?

Matrix-Zhang, 2020-08-13:
> @Matrix-Zhang I need more information on it can you provide me the device logs with the error?

where's the device logs? I didn't see much information from the console output

remonh87, 2020-08-14:
the flutter console is always a bit limited, logcat output provides more data. For me it is impossible to debug this based on the current information. It would be helpful if you could provide some more background

remonh87, 2020-09-06:
Closed the issue due to inactivity feel free to reopen in case you still encounter issues. Please add some additional error messages in that case so we can help solving the issue.

closed 2020-09-06 by remonh87