PhilipsHue/flutter_reactive_ble #102: About value type for the write question

yihuishou, 2020-08-24:
**Is your feature request related to a problem? (please describe)**
``` dart

final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: characteristicUuid, deviceId: foundDeviceId);
reactiveBleClient.writeCharacteristicWithoutResponse(characteristic, value: [0x00]);

```

The parameter must be type of int hex?  

**Describe the solution you'd like**

The result obtained by bit operation

for example：

`int result = ( 1 << 7 ) | 100; `

result is 228;

The hexadecimal string is obtained by using the method `toRadixString(16)` , get `e4` String

According to the document, the parameter should be `0xe4`

But I can't find a way to convert 'e4' to `0xe4`

**Describe alternatives you've considered**

Maybe hexadecimal strings can be supported?

werediver, 2020-08-24:
@yihuishou You are lacking the basic understanding of data types in programming languages and/or computer systems in general. A number is a number regardless of the numerical system it is displayed or written in. As long as your variable has a numerical type (such as `int`) and is bound to the range of 0–255 (fits into one byte) you can send it to a device.

A string (text) is different. Even if text looks like a number, it's still text. Sometimes that's what you need, but you should very well understand which kind of data is expected by a device to correctly communicate to it.

yihuishou, 2020-08-24:
In other words, even if the interface in the document requires the use of hexadecimal int, the int passed in decimal can work normally, right?
Thank you for your answer. 😃

werediver, 2020-08-24:
> the int passed in decimal can work normally, right?

This is right.

> the interface in the document requires the use of hexadecimal int

This is not. An example is not a requirement. The function signature is, and it requires a list of integers. In Dart you can write an integer in decimal or hexadecimal form or compute it based on other variables and literals written in decimal and hexadecimal forms. It's quite safe to generalize this principle to most of programming languages.

I hope that resolves your question.

closed 2020-08-24 by werediver