PhilipsHue/flutter_reactive_ble #105: Device name not updating on Android

soyangel, 2020-09-15:
**Describe the bug**
BLE Devices changes its name but the name in not changed on the data returned on scan. 

**To Reproduce**
Try scanning a device that changes its name on the advertising data but then on the scan results the name is not changed.

To fix this you can change ReactiveBleClient.kt:

```
                 .map { result ->
-                    ScanInfo(result.bleDevice.macAddress, result.bleDevice.name
+                    ScanInfo(result.bleDevice.macAddress, result.scanRecord.deviceName
                             ?: "",
                             result.rssi,
                             result.scanRecord.serviceData.mapKeys { it.key.uuid },
```

**Expected behavior**
Device name changes on scan results.

**Smartphone / tablet**
 - Device: Pixel 3a
 - OS: Android 10

**Peripheral device**
- Vendor, model: Custom
- Does it run a custom firmware / software: yes

**Additional context**

werediver, 2020-09-16:
Relates to #83 (same issue on iOS, fixed).

remonh87, 2020-09-18:
Will be available in coming release thanks for reporting!

PR #106 (2020-09-18): use scanrecord instead of ble devie
Fixes #105
closed 2020-09-18 by remonh87