PhilipsHue/flutter_reactive_ble #112: Support for new Desktop platforms.

JoeGilkey, 2020-10-04:
**Is your feature request related to a problem? (please describe)**
It would be very nice to have support for the new desktop platforms that Flutter has added.

**Describe the solution you'd like**
I would like to share code across an Android/iOS solution with a desktop solution. Each of the desktop platforms have ways to use BLE. For the Windows platform, it appears there is a feature request in https://github.com/timsneath/win32 to add bluetooth support.

**Describe alternatives you've considered**
I haven't found any alternatives yet for Flutter. Xamarin does have a cross platform library (https://github.com/nexussays/ble.net) but it doesn't support Linux or Mac.

**Additional context**
I'm currently using this library in a project to talk to an OBD-2 device that has BLE connectivity. It would be nice to have the same app modified for desktop so users can use their laptops. I've had numerous requests for Chromebook support, so the Linux desktop is very appealing.

remonh87, 2020-10-04:
@JoeGilkey  That is a nice feature request but not something we have on our roadmap on the short-term. However if you or someone else wants to provide support for desktop platforms we are more than happy to assist.

werediver, 2020-10-05:
In fact, our iOS implementation should be suitable (maybe with minor amendments) for macOS, but that's it. Neither Linux, nor Windows are supported or planned, as Remon mentioned.

Perhaps, the new [federated plug-in architecture](https://flutter.dev/docs/development/packages-and-plugins/developing-packages#federated-plugins) can be of use here, but it's not on our roadmap either.

ekuleshov, 2021-02-28:
So far this project demonstrated great activity in regards to maintaining code, providing fixes and applying community-provided pull requests. However lack of plans regarding support for desktop platforms is very discouraging. Especially macos and then windows platforms.

remonh87, 2021-02-28:
@ekuleshov I understand your request,  me and @werediver are actively maintaining iOS and Android which is already quite a big task that we do in our freetime. At this moment we do not have any bandwith to incorporate support for more platforms, however, if the community is willing to take on this task we are more then happy to support in reviewing prs and helping ppl get started

lukepighetti, 2021-03-07:
May I recommend adopting the new federated plugin architecture so that this repo is ready to start accepting work in these areas?

remonh87, 2021-03-07:
Yess it is on the road map #128 . Null-safety migration is almost done so I think moving to a federated structure would be a nice next step.

remonh87, 2021-03-28:
We have split this issue into more platform specific tickets and added a label to it. I will close this issue now

https://github.com/PhilipsHue/flutter_reactive_ble/issues?q=is%3Aissue+is%3Aopen+label%3A%22desktop+support%22

closed 2021-03-28 by remonh87