PhilipsHue/flutter_reactive_ble #113: How to stop Scanning to devices?  (Or how to stop the scanForDevices stream?)

0100101101001011, 2020-10-05:
I'm quite new to flutter. I'm wondering how to stop scanning for peripheral after getting a list of devices?  

**To Reproduce**
`FlutterReactiveBle _ble = FlutterReactiveBle();

    _ble.statusStream.listen((status) {
      //code for handling status update
      print(status);
    });

    print(_ble.status);

    void startScan() {
      _ble.scanForDevices(withServices: [], scanMode: ScanMode.lowLatency, requireLocationServicesEnabled: true).listen((device) {
        //code for handling results
        print(
            'ID:' + device.id + 'Name:' + device.name
        );
        print(device.rssi);
      }, onError: (handleError) {
        //code for handling error
        print(handleError);
      });
    }`

werediver, 2020-10-05:
To stop a scan (as well as to close an established connection) you need to unsubscribe from the corresponding stream.

You're creating a "subscription" object by calling [listen(⋯)](https://api.flutter.dev/flutter/dart-async/Stream/listen.html) method. Later you can [cancel()](https://api.flutter.dev/flutter/dart-async/StreamSubscription/cancel.html) this subscription.

0100101101001011, 2020-10-05:
Thank you very much! BTW, this library is amazing!

closed 2020-10-07 by werediver