PhilipsHue/flutter_reactive_ble #114: "Scanning parameters are not set" on Hot Restart

xvrh, 2020-10-05:
**Describe the bug**
On Android, I consistently have a failure on the first `scanForDevices` after a Hot Restart.

**To Reproduce**
Steps to reproduce the behavior:

This program have a single button to start a bluetooth scan
```
import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';

void main() => runApp(MaterialApp(
      home: Scaffold(
        body: BluetoothApp(),
      ),
    ));

final _client = FlutterReactiveBle();

class BluetoothApp extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return SafeArea(
      child: RaisedButton(
        onPressed: () {
          _client.scanForDevices(
              withServices: [], scanMode: ScanMode.lowLatency).listen((device) {
            print('Found ${device.name}');
          }, onError: (e) {
            print('Scan error $e');
          }, cancelOnError: true);
        },
        child: Text('Start scan'),
      ),
    );
  }
}
```
- Run the program and press "Start scan"
- During the scan, execute a Hot Restart
- After the application restart, press "Bluetooth scan" again
- It fails with `Scan error Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scanning parameters are not set")`
- Press a second time, it works now.

**Expected behavior**
It would improve developer experience if the library was more resilient to this case.

**Smartphone / tablet**
I can consistently reproduce on this 3 devices:
Motorola G5 (Android 8.1.0), Huawei P9 lite (Android 7.0), Nokia android One (Android 9)

Package version: 2.5.0

**Peripheral device**
NA

**Additional context**
This error message has already been reported here: https://github.com/PhilipsHue/flutter_reactive_ble/issues/87 but I don't known if this is the same issue.

remonh87, 2020-10-07:
@xvrh good catch I fixed the issue and it should be available in release 2.5.2

closed 2020-10-07 by remonh87

xvrh, 2020-10-07:
Excellent. Thanks 👍

ened, 2022-02-17:
This may have regressed in the latest versions, @remonh87

CPMoviePHil, 2022-07-07:
face the same issue here 
flutter_reactive_ble: ^5.0.2
android 12
pixel 4(real device)

sdale511, 2024-02-09:
same issue here with android 12 pixel 3a
flutter_reactive_ble: ^5.0.3

HugoHeneault, 2024-02-13:
Same same :')

PR #117 (2020-10-07): Fix issue devicescan android on hot restart
Fixes #114