PhilipsHue/flutter_reactive_ble #115: Conflict between Firebase Packages and Flutter_reactive_blue

plyrek, 2020-10-06:
The Firebase Packages:
  firebase_core: ^0.5.0
  firebase_auth: ^0.18.0+1
  cloud_firestore: ^0.14.0+2
recently updated to use protobuf-javalite - 3.110.jar which conflicts with the jar file protobuf-lite-3.0.1.jar used in your package making any app which uses Firebase incompatible with your package. You are not alone, several ble packages use this jar and are incompatible, Your package however, was very recently updated and seems active and I thought you would like to know.

**To Reproduce**
Steps to reproduce the behavior:
Add the following to pubspec.yaml and try to build an android app
firebase_core: ^0.5.0
firebase_auth: ^0.18.0+1
cloud_firestore: ^0.14.0+2
flutter_reactive_ble: ^2.5.1

**Expected behavior**
A clean build to an Android device or emulator

**Actual behavior**
Build fails with the following errors
1 exception was raised by workers:
  java.lang.RuntimeException: Duplicate class com.google.protobuf.AbstractMessageLite found in modules protobuf-javalite-3.11.0.jar (com.google.protobuf:protobuf-javalite:3.11.0) and protobuf-lite-3.0.1.jar (com.google.protobuf:protobuf-lite:3.0.1)

remonh87, 2020-10-07:
@plyrek  thanks for reporting the issue. I couldn't reproduce it by using the steps you supplied. What I did was following: 

1. Create flutter sample project 
2. Integrate the dependencies like you said.
3. Build it on a google pixel2 device (which was successful).

I did however upgrade the protobuf dependencies and will integrate them soon in master. Can you test it with your project again? Else I need a bit more info: 

- output of `flutter doctor`
- build.gradle file where you declare the depedencies. If that is problematic please let me know which gradle version, external depedencies and android tools you are using

plyrek, 2020-10-07:
Thank you so much for your quick action on this issue. The updating of protobuf dependencies did the trick! It is contributors like you that make the Flutter community such a great one!

archanpaul, 2020-11-11:
The problem persists with following : 

**pubspec.yaml** 

firebase_core: ^0.5.2
firebase_auth: ^0.18.3
cloud_firestore: ^0.14.3
firebase_remote_config: ^0.4.2
firebase_admob: ^0.10.2
flutter_reactive_ble: ^2.6.1+1

**Error** 

> 1 exception was raised by workers:
  java.lang.RuntimeException: Duplicate class com.google.protobuf.AbstractMessageLite found in modules jetified-protobuf-javalite-3.11.0.jar (com.google.protobuf:protobuf-javalite:3.11.0) and jetified-protobuf-lite-3.0.1.jar (com.google.protobuf:protobuf-lite:3.0.1)

Considering lot of people use the firebase, it will be good if flutter_reactive_ble remain compatible with protobuf library of firebase plugins.

remonh87, 2020-11-11:
@archanpaul  I will reopen it because of the visibility for us. We have btw a version with the latest protobuf: `2.6.1` but unfortunately we had to drop it because of the firebase info you mentioned.

archanpaul, 2020-11-11:
@remonh87 No I have not declared firebase BOM in gradle.

remonh87, 2020-11-16:
This issue is fixed in version 2.7.1

closed 2020-11-16 by remonh87

archanpaul, 2020-11-17:
Thank you @remonh87 for the fix. Working with following (all latest as of today) :
  firebase_core: ^0.5.2
  firebase_auth: ^0.18.3
  cloud_firestore: ^0.14.3
  firebase_remote_config: ^0.4.2
  flutter_reactive_ble: ^2.7.1

PR #116 (2020-10-07): upgrade android dependencies
Fixes #115