PhilipsHue/flutter_reactive_ble #118: no connectedDeviceStream updates after connectToDevice subscription cancel

d-wolf, 2020-10-07:
**Describe the bug**
If i listen to the ```connectedDeviceStream``` of a ```FlutterReactiveBle```instance, connect to a peripheral with ```connectToDevice``` and cancel the subscription to disconnect the peripheral later, the ```connectedDeviceStream``` notifies no ```ConnectionStateUpdate``` with the states ```DeviceConnectionState.disconnecting``` & ```DeviceConnectionState.disconnected```.

**To Reproduce**
Steps to reproduce the behavior:
```Dart
...

final _ble = FlutterReactiveBle();
_stateUpdateSubscription = _ble.connectedDeviceStream.listen(_onConnectDevice);
_connectionStateUpdateSubscription = _ble.connectToDevice(id: myPeripheral.Id).listen(null);
await _connectionStateUpdateSubscription?.cancel();
...

void _onConnectDevice(ConnectionStateUpdate state) {
    _logger.i(state.connectionState);
  }
```

**Expected behavior**
The ```connectedDeviceStream```should update twice when i disconnect a peripheral manually. The first update should be with the state ```DeviceConnectionState.disconnecting```, the second with ```DeviceConnectionState.disconnected```

**Smartphone / tablet**
 - Device: Nokia 7.1
 - OS: Android 10
 - Package version: 2.5.1

**Peripheral device**
- Vendor, model: BM832, nRF52832 Bluetooth 5 Module
- Does it run a custom firmware / software: yes

**Additional context**
Shouldn't the  ```connectedDeviceStream``` get also updated after i manually disconnect a device?

werediver, 2020-10-09:
Is it reproducible on Android only? (= did anybody try the same on iOS?)

remonh87, 2020-10-09:
@werediver  we do disconnect when a user cancels the stream. So it makes sense that the disconnect and disconnecting is not received. We can choose to change the behaviour but that would require quite some thinking and refactoring. Let's maybe discuss this on Monday.

@d-wolf  in the example app we solve it by adding a update to the StreamController :

```dart

 Future<void> disconnect(String deviceId) async {
    if (_connection != null) {
      try {
        await _connection.cancel();
      } on Exception catch (e, _) {
        print("Error disconnecting from a device: $e");
      } finally {
        // Since [_connection] subscription is terminated, the "disconnected" state cannot be received and propagated
        _deviceConnectionController.add(
          ConnectionStateUpdate(
            deviceId: deviceId,
            connectionState: DeviceConnectionState.disconnected,
            failure: null,
          ),
        );
      }
    }
  }

```
This is also how we solve it in our ble apps.

werediver, 2020-10-09:
@remonh87 You're talking about the "filtered" connection status stream _returned by the "connect" method_. The OP, I believe, is talking about the _multiplexed_ (non-filtered-all-devices) connection status stream. There I'd expect the disconnection events.

remonh87, 2020-10-09:
@werediver  my bad indeed was about the multiplexed one. I noticed on iOS we emit disconnected on Android we do not do this which is suspicious. So it is a bug on the Android implementation.

remonh87, 2020-10-11:
@d-wolf issue is fixed in version 2.5.3 thanks for reporting

closed 2020-10-11 by remonh87

d-wolf, 2020-10-12:
@remonh87 Thank you for the fast support!

PR #121 (2020-10-11): emit disconnected state update when device gets disconnected
Fixes #118