PhilipsHue/flutter_reactive_ble #119: BLE is being unauthorized in Android

pachofisico, 2020-10-07:
**Describe the bug**
I am receiving this message: "Authorize the FlutterReactiveBle example app to use Bluetooth and location"

**To Reproduce**
Android 7.

Steps to reproduce the behavior:
1. Download flutter_reactive_ble.
2. Debug in Android Studio, or flutter run in VS Code.

Result: 
Unauthorized -  Authorize the FlutterReactiveBle example app to use Bluetooth and location

**Smartphone / tablet**
Samsung Galaxy Note 5

remonh87, 2020-10-08:
@pachofisico  this is because you need to give the reactive ble app permissions by going to settings => apps => select the app => permissions => enable location. 

We are not handling the permission programatically since we want to allow consumers to implement handling the permissions themselves.  Normally a package like https://pub.dev/packages/permission_handler would do the trick for you

remonh87, 2020-10-11:
Closed the issue now feel free to open a new issue in case you have any questions or reopen the existing one if you feel like your question is not answered

closed 2020-10-11 by remonh87

saibotma, 2021-09-09:
Would be great if you could add a quick note about this to the README.

menari123, 2021-11-03:
Information like this should be in the READ.me file, as it is essential for your project to work.

remonh87, 2021-11-03:
I normally prefer to document this inside the code so it becomes readable when you consume this method in your IDEA. I added this to the [docs](https://github.com/PhilipsHue/flutter_reactive_ble/blob/14a5c1b51b3b17711155d0a4ab564d13e8d18c47/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L218). Let me know if this is clear enough or else I will spend a bit more time explaining it there

menari123, 2021-11-03:
Thinking about who wants to study the code is a great idea, but it can end up giving a certain headache to those who just want to clone the repository and run the project without having to read all the .dart files.
Great job and good luck

saibotma, 2021-11-07:
> I normally prefer to document this inside the code so it becomes readable when you consume this method in your IDEA. I added this to the [...]

I'm sorry, but I can't find where you added the documentation. Can you please provide another link?

BradKwon, 2021-12-16:
Agree. Instead of saying that in the code, it might be better to change the text message more precisely by adding "**Please manually** authorize the FlutterReactiveBle example..." if it can't be written in readme I guess.

truedat101, 2023-10-10:
The problems that seem to arise on this in a "modern" android are related to the "Scan For Nearby Devices" permission which isn't obvious how to set that on android.  It is tied to Wifi apparently so I can see how this is left as an exercise, but it would be just a bit more streamlined to have an example of how to handle this cleanly in a manifest file for modern android devices and maybe the example can have a pseudo code comment regarding *this is where you should check permissions, this is where you should open settings if on android , etc.