PhilipsHue/flutter_reactive_ble #122: Max number of concurrent BLE connections? And how to send data to mutiple BLE devices at the same time?

0100101101001011, 2020-10-11:
**Describe**
I am wondering that what is the maximum number of active BLE connections that can be achieved with stability? And how to send the same data to multiple devices at the same time? Do you subscribe to characteristics one by one or it can be done by some sort of device list?

Now I can send data to a single device 👍 
**To Reproduce**
`void subscribeToCharacteristic() async{
  final characteristic = QualifiedCharacteristic(serviceId: Uuid.parse('6E400001-B5A3-F393-E0A9-E50E24DCCA9E'), characteristicId: Uuid.parse('6E400002-B5A3-F393-E0A9-E50E24DCCA9E'), deviceId: '4490647D-0F3C-868B-F80B-00957ACFB4CC');
  //final response = await _ble.readCharacteristic(characteristic);
  final response = await _ble.writeCharacteristicWithResponse(characteristic, value: utf8.encode('sendData'));
  //print(response);
}`

werediver, 2020-10-11:
The number of _connections_ a particular combination of hardware and software (BLE stack) can reliably maintain varies quite a bit between different vendors and makes, but can be as little as 2–3 (on _some_ Android devices) or about 11+ on iPhones and _some_ Android devices. This is pretty much unpredictable, unless you have test data.

A single _peripheral_ (a smartwatch) can "broadcast" data via _notifications_ (not _indications_! _notifications_ do not require acknowledgement and can be received by multiple _central_ nodes, _indications_ do require and are addressed to a particular _central_ node).

A _central_ device (a smartphone in a typical configuration) cannot send data to multiple _peripherals_. A smartphone can work in _peripheral_ mode, though. Then it can send notifications, but only _central_ nodes will receive it.

This library does not support operation in _peripheral_ mode and arguably shouldn't, because its complexity is high enough already.

0100101101001011, 2020-10-11:
Thank you so much for the quick reply! 

If I want to send data to multiple peripherals in a very short amount of time(instead of at the exact same time), is it possible?

0100101101001011, 2020-10-11:
Can I do something like this?

```
void sendDataToAllDevices(data) {
     deviceLists.forEach((device) {
        final characteristic = QualifiedCharacteristic(serviceId: device[serviceUuid], characteristicId: device[characteristicUuid], deviceId: device[deviceID]);
reactiveBleClient.writeCharacteristicWithoutResponse(characteristic, value: [0x00]);
      });
    }
```

0100101101001011, 2020-10-11:
> The number of _connections_ a particular combination of hardware and software (BLE stack) can reliably maintain varies quite a bit between different vendors and makes, but can be as little as 2–3 (on _some_ Android devices) or about 11+ on iPhones and _some_ Android devices. Bluetooth Classic connections are usually count together with BLE connections. This is pretty much unpredictable, unless you have test data.
> 
> Another type of a limit is how much _subscriptions_ you can have in total. On vanilla Android there is a hard-coded limit of something _under 20_ subscriptions, but that may be changed by a vendor. I am not aware of a similar limit on iOS.
> 
> A single _peripheral_ (a smartwatch) can "broadcast" data via _notifications_ (not _indications_! _notifications_ do not require acknowledgement and can be received by multiple _central_ nodes, _indications_ do require and are addressed to a particular _central_ node).
> 
> A _central_ device (a smartphone in a typical configuration) cannot send data to multiple _peripherals_. A smartphone can work in _peripheral_ mode, though. Then it can send notifications, but only _central_ nodes will receive it.
> 
> This library does not support operation in _peripheral_ mode and arguably shouldn't, because its complexity is high enough already.

werediver, 2020-10-11:
> Can I do something like this?
> 
> ```dart
> void sendDataToAllDevices(data) {
>     deviceLists.forEach((device) {
>         final characteristic = QualifiedCharacteristic(serviceId: device[serviceUuid], characteristicId: device[characteristicUuid], deviceId: device[deviceID]);
>         reactiveBleClient.writeCharacteristicWithoutResponse(characteristic, value: utf8.encode(data));
>     });
> }

You can iterate over a list of devices and send data to them. You may consider using write-without-response, if you value the speed (latency) over reliability (in practice, though, there should be no issues with reliability anyway). But you have to _await_ for your calls, something you're _not_ doing in this example. Try the normal for-each and await.

0100101101001011, 2020-10-14:
Thanks for the help!

closed 2020-10-14 by 0100101101001011