PhilipsHue/flutter_reactive_ble #123: How to get connectionState without creating a stream?

0100101101001011, 2020-10-14:
**Describe the problem**
Is there a way to get the **connectionState** status without creating a stream, just like doing **_ble.status**.

I am not sure which is the best way to implement this. I have several screens in my app, so I am creating a **connectToDevice** stream when the app loads on the home screen widget. But I also need to access the connectionState status on other screens as well to know the device is still connected. So I am calling the **connectToDevice** stream again on other screens which creates multiple streams for the same device. 

**To Reproduce**
On Screen One:
```
class ScreenOne extends StatelessWidget {

  @override
  Widget build(BuildContext context) {

    _ble.connectToDevice(
      id: theDeviceID,
      connectionTimeout: const Duration(seconds: 3),
    ).listen((connectionState) {
      print(connectionState);
      //Do something if the state changes
    }, onError: (Object error) {
      print(error);
    });

    return Container();
  }
}
```
On Screen Two:
```
class ScreenTwo extends StatelessWidget {

  @override
  Widget build(BuildContext context) {

    _ble.connectToDevice(
      id: theDeviceID, //Same device
      connectionTimeout: const Duration(seconds: 3),
    ).listen((connectionState) {
      print(connectionState);
      //Do something if the state changes
    }, onError: (Object error) {
      print(error);
    });

    return Container();
  }
}
```

remonh87, 2020-10-15:
@0100101101001011  there are many ways to solve this problem.  You can have a look to our example app where we use wrapper objects in combination with [provider](https://pub.dev/packages/provider). See our implementation: [example](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/main.dart). 

In general I would create a simple wrapper that updates your appstate and then choose a way you like to add it to the screen.

0100101101001011, 2020-10-15:
Thank you! I achieved it by updating the connectionState to a local sembast database.

closed 2020-10-15 by 0100101101001011