PhilipsHue/flutter_reactive_ble #125: Class DiscoveredService isn't exported

smafre, 2020-10-16:
I'm trying to create a function that discovers services for a deviceId and returns the result. It's really just a small change from the function in the example project.
I'm changing
`Future<void> discoverServices(String deviceId) async {
    ...
}`
to
`Future<List<DiscoveredService>> discoverServices(String deviceId) async {
    ...
}`

I'm having troubles importing/using the type `DiscoveredService` since the file `discovered_services.dart` isn't exported in `flutter_reactive_ble.dart` in the library.

Is this intentional or could we add `discovered_services.dart` to the list of exported files in `flutter_reactive_ble.dart`? If so I'm happy to make a PR with that small change.

closed 2020-10-19 by remonh87

remonh87, 2020-10-20:
@smafre thanks for reporting the issue. We just shipped version 2.6.0 which should include a fix for your issue

smafre, 2020-10-20:
@remonh87 Thanks a lot for resolving this so quick, and for an excellent plugin so far!

PR #126 (2020-10-19): Export `DiscoveredService` class
Fixes #125