PhilipsHue/flutter_reactive_ble #130: Placeholder migrate lib to null safety

remonh87, 2020-10-23:
**Is your feature request related to a problem? (please describe)**
End of the year the Flutter will introduce null safety in beta: https://bit.ly/3kp6N4r .  At this point it is just in tech preview but once it gets integrated in Beta I think we should try out the migration tooling from the Flutter team

**Describe the solution you'd like**
Migrated library that is ready for null-safety once it gets into stable.

farr64, 2020-10-23:
This is a wonderful development and I'm happy to see that you are on board. Thanks and go for it.

remonh87, 2020-11-22:
Did investigation and we still need to wait for 3 dependencies before they can be migrated:

- [ ]  [functional_data](https://github.com/spkersten/dart_functional_data/issues/21)
- [ ] [mockito](https://github.com/dart-lang/mockito/issues/298)

remonh87, 2021-02-16:
first version is launched. We created seperate tickets and linked it to the null safety milestone. This can be closed.

closed 2021-02-16 by remonh87