PhilipsHue/flutter_reactive_ble #131: Example app crashing in Release scheme

dawidb-appdate, 2020-10-23:
**Describe the bug**
hey, when building android app using standard command 'flutter build apk --release' package is created, but when loaded to actual phone it crashes every time. I was managed to figure out that it's the very first line is crashing 'final _ble = FlutterReactiveBle();', so basically loading/initializing the library.
When running in debug on the same device it works well. 
Have you guys had similar problems? What could be causing the problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Build release scheme with command 'flutter build apk --release'
2. Load the app to actual device 
3. Run app and see it crashing

**Smartphone / tablet**
 - Device: Mi MIX 2s
 - Android 9
 - 
**Additional context**
Android studio 4.0.2
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 1.22.1, on Mac OS X 10.15.7 19H2, locale en-PL)
[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 12.0.1)
[✓] Android Studio (version 4.0)

Log from phone:
`d.b.z.d: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | java.lang.RuntimeException: Field status_ for c.d.a.a.c not found. Known fields are [private static final c.d.a.a.c c.d.a.a.c.e, private static volatile c.a.a.b1 c.d.a.a.c.f]
    at d.b.b0.b.a$h.a(Unknown Source:2)
    at d.b.b0.b.a$h.f(Unknown Source:2)
    at d.b.b0.d.h.b(Unknown Source:13)
    at d.b.b0.d.h.e(Unknown Source:25)
    at d.b.b0.e.e.i0$a.l(Unknown Source:47)
    at d.b.b0.e.e.i0$a.run(Unknown Source:8)
    at d.b.x.b.b$b.run(Unknown Source:2)
    at android.os.Handler.handleCallback(Handler.java:873)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loop(Looper.java:201)
    at android.app.ActivityThread.main(ActivityThread.java:6806)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:547)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:873)
Caused by: java.lang.RuntimeException: Field status_ for c.d.a.a.c not found. Known fields are [private static final c.d.a.a.c c.d.a.a.c.e, private static volatile c.a.a.b1 c.d.a.a.c.f]`

remonh87, 2020-10-23:
@dawidb-appdate I need a bit more background on this  do you use firebase or google play services in the app?  Also which version of the library are you using? Can you try it out with version 2.5.1 and 2.6.1 ?

dawidb-appdate, 2020-10-23:
@remonh87 i was using the latest release 2.6.0 and 2.5.3 - both the same issue (debug build works correctly on the same device)
as for firebase/google play - i didn't add anything to the app (only added signingConfigs in app/build.gradle so i can sign it with the key). I'm downloading the latest, updating app/build.gradle with signing key portion and building apk from terminal. that's all.

remonh87, 2020-10-24:
@dawidb-appdate  it is not a signing problem but I suspect something related to building releases (can be pro guard for example). Can you try it out with the versions I described in my previous comment? (2.5.1 and 2.6.1). Just want to excluded that the protobuf upgrade is related to the problem.

dawidb-appdate, 2020-10-24:
@remonh87 i just wanted to stressed out that nothing was changed in the code ;)
2.5.1 - builds properly (no errors which are described below) and release version is running smoothly on the device (so i will stick with this one for now)
2.6.1/2.6.0/2.5.3 has the same issue where it's not running properly with release scheme. But i noticed something which may help. 
**When building release version i got the error:**

`
FAILURE: Build failed with an exception.                                
                                                                         What went wrong:                                                      
Execution failed for task :app:lintVitalRelease.                      
> Could not resolve all artifacts for configuration ':app:debugRuntimeClasspath'.
   > Failed to transform libs.jar to match attributes {artifactType=processed-jar, org.gradle.libraryelements=jar, org.gradle.usage=java-runtime}.
      > Execution failed for JetifyTransform: /Users/dawid/Git/flutter_reactive_ble/example/build/app/intermediates/flutter/debug/libs.jar.
         > Transform's input file does not exist: /Users/dawid/Git/flutter_reactive_ble/example/build/app/intermediates/flutter/debug/libs.jar. (See https://issuetracker.google.com/issues/158753935)`

 **So you need to build debug first _flutter build apk --debug_, but then you get** 
`FAILURE: Build failed with an exception.                                
                                                                        What went wrong:                                                      
Execution failed for task ':app:lintVitalRelease'.                      
> Could not resolve all artifacts for configuration ':app:profileRuntimeClasspath'.
   > Failed to transform libs.jar to match attributes {artifactType=processed-jar, org.gradle.libraryelements=jar, org.gradle.usage=java-runtime}.
      > Execution failed for JetifyTransform: /Users/dawid/Git/flutter_reactive_ble/example/build/app/intermediates/flutter/profile/libs.jar.
         > Transform's input file does not exist: /Users/dawid/Git/flutter_reactive_ble/example/build/app/intermediates/flutter/profile/libs.jar. (See https://issuetracker.google.com/issues/158753935)`

**So you need to build flutter build apk --profile and you'll get error-free release build with _flutter build apk --release_**
**But when you install 2.6.1/2.6.0/2.5.3 it will crash on the device**

remonh87, 2020-10-25:
I think there is still a collision with a Firebase plugin and the protobuf upgrade that we did. I had the same issue with firebase remote config and could reproduce it on release builds .I will revert the upgrade and will launch a new version, probably tomorrow.

remonh87, 2020-10-26:
@dawidb-appdate fix for this is avaialble in version 2.6.1+1 .

closed 2020-10-26 by remonh87

omarkamal, 2020-11-18:
Hey @dawidb-appdate, how'd you get the log from the phone? I'm having the exact same issue, where it's crashing in release mode only.

remonh87, 2020-11-18:
@omarkamal  since latest update you need to add our library to the the ProGuard exceptions. For some reason the generated protobuf files are stripped when we use the latest one. See our readme how to do this. Let me know if this works for you

omarkamal, 2020-11-18:
Thanks for your swift response! Will check it out and share my findings here

omarkamal, 2020-11-18:
I don't have a proguard rules file, I haven't enabled it. Proguard seems to have even disappeared from the flutter official docs deployment page for android.

remonh87, 2020-11-18:
By default Android uses shrinking of the code. See for more info https://developer.android.com/studio/build/shrink-code. 

Check you release config in `build.gradle` and you probably have something like this: 

```
buildTypes {
        release {
            signingConfig <signing stuf>
            debuggable false
            minifyEnabled true
            shrinkResources true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}
```

The last line is very important if you do not have it I will suggest you add `proguard-rules.pro'` to your app folder and add the line: `-keep class com.signify.hue.** { *; }` . This should help. The reason why it does work on debug and not on release in general has to do with this.

omarkamal, 2020-11-18:
Thanks @remonh87, I did that and it worked fine! Cheers, and thanks for your wonderful library.