PhilipsHue/flutter_reactive_ble #132: Any experiences with "dependent dylib not found" (which has been also reported elsewhere)?

farr64, 2020-10-25:
This is neither a bug report nor a feature request. I ran into this and, searching the forums, I see that several other people have faced the same issue.

To see a few examples, search for: `DYLD, dependent dylib '@rpath/Flutter.framework/Flutter' not found`

I'm simply curious to see if any of you have also seen this.

Please see if you can reproduce this issue by following these steps:

With the clean and unchanged version from your repository (with the minimally required changes for "Automatically manage signing" in Xcode), use Android Studio to run your Example app on a physical iPhone.

Running the app (from Android Studio) causes the app to crash. I have attached the log file from Xcode.

However, running the corresponding Runner from Xcode works perfectly.

So, there is no problem running the app on a physical iPhone using this kind of "manual" mode.

The challenge is that we can't use Hot Reload on the iPhone and that is a real pity.

I have noticed that this happens "some times" but not "other times". So, the behavior is frustratingly inconsistent.

Any ideas or suggestions?

Thanks.

[dependent dylib not found.txt](https://github.com/PhilipsHue/flutter_reactive_ble/files/5434143/dependent.dylib.not.found.txt)

farr64, 2020-10-25:
Aha! I just tried on another iPhone and it worked as expected. 

I have Xcode 12.1 on my Mac.

The iPhone that fails has iOS 13.4.1

The iPhone that succeeds has iOS 14.0.1

So, even before running more tests, it appears that there might be some inconsistencies between various versions of Xcode and iOS.

I would appreciate any comments so we can all learn, to avoid having to scratch our heads unnecessarily. Thanks.

werediver, 2020-10-26:
Have you tried removing the whole `ios/Flutter/` directory in your app package (to let the toolchain reconstruct it)? If you have no customization inside `ios/`, I'd suggest to delete the whole `ios/` instead (the toolchain will regenerate it). Alternatively, you can try re-integrating CocoaPods.

Try to upgrade your Flutter SDK to the latest stable or even Beta, if it's acceptable.

There was a somewhat similar issue with iOS 13.3 (flutter/flutter/issues/49504), but it's claimed to be resolved in iOS 14, which does not match your observations.

farr64, 2020-10-26:
Thanks for the tips,@werediver. 

I'll follow your advice in the next day or so and I'll report back.

farr64, 2020-10-27:
The iOS chain is, apparently, very sensitive to certain mismatches. For instance, "The application crashes on launch" is among the Known Issues for Xcode 12.2 Beta 3 for certain mismatches:
https://developer.apple.com/documentation/xcode-release-notes/xcode-12_2-beta-release-notes/

By doing things "just so" (context-depending, I guess) I can manage to build and run,.

No big deal. Part of being on the bleeding edge, I suppose.

Thanks for the suggestions.

closed 2020-10-27 by farr64