PhilipsHue/flutter_reactive_ble #134: flutterApp/build/flutter_reactive_ble/generated/source/proto/debug/javalite/com/signify/hue/flutterreactiveble/ProtobufModel.java uses unchecked or unsafe operations.

0100101101001011, 2020-10-27:
When building for Andriod:

```
Note: /Users/user/AndroidStudioProjects/FlutterProject1/build/flutter_reactive_ble/generated/source/proto/debug/javalite/com/signify/hue/flutterreactiveble/ProtobufModel.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.

FAILURE: Build failed with an exception.
```

remonh87, 2020-10-27:
@0100101101001011  we are aware of this problem and logged a ticket for the Firebase team: https://github.com/FirebaseExtended/flutterfire/issues/3847 . Unfortunately I had to revert the protobuf upgrade due to this issue but you can still use version `2.6.1` which has the upgraded protobuf

0100101101001011, 2020-10-27:
Thank you for the quick reply! I am using 2.6.1+1, but I am still getting the error. Should I change anything in my build.gradle file? 
[image]

remonh87, 2020-10-27:
you need version `2.6.1` without the plus or else `2.5.3`

0100101101001011, 2020-10-27:
Thank you! Built successfully by using 2.6.1!

closed 2020-11-11 by remonh87