PhilipsHue/flutter_reactive_ble #135: Question: is connectToAdvertisingDevice() method android only?

iJamesPHP, 2020-11-01:
Is the connectToAdvertisingDevice() method android only? It hangs for me and never connects on iOS. Does it require you to stop scanning before using it?

werediver, 2020-11-02:
`connectToAdvertisingDevice(⋯)` wasn't designed as Android specific. It should work on iOS, but you don't need it there, because iOS is very good at establishing connections (and handling BLE in general). I'd recommend to check if the program is running on Android before using this method.

If I remember correctly, the scanning should be stopped before you invoke it (regardless of the platform), indeed. @remonh87 could you comment on this?

remonh87, 2020-11-02:
yes it is advised to stop scanning before invoking it else especially on Android it can lead to flaky results.

closed 2020-11-05 by werediver