PhilipsHue/flutter_reactive_ble #136: Question: reconnection to device after reboot.

iJamesPHP, 2020-11-02:
How should connecting to a device after it reboots be handled best? My solution of just connecting in the disconnect listener works fine on iOS. However, on Android it doesn't reconnect quick enough to capture the data I need (within 10 sec of reboot). Is there something I'm missing? Is stopping the scan crucial to anything working well on Android?

remonh87, 2020-11-02:
@iJamesPHP connecting while scanning on Android is not recommended. You should first stop scanning and then try to connect. For more info about ble on Android and some more behaviour see: https://bit.ly/2HXddK1. Btw our library already works around a lot of these known issues.

Regarding the time, 10 seconds is quite long can you provide me the code that you are using?

remonh87, 2020-11-07:
close the ticket for now feel free to reopen

closed 2020-11-07 by remonh87